moveit/moveit #928: Switching default branches to melodic-devel

davetcoleman, 2018-05-31:
I am going to switch all the ros-planning MoveIt! repos to ``melodic-devel`` as default (where the branch exists) unless anyone would like to discuss first. This is what I typically do after the distribution (Meldoic) is officially released.

Of course you can still make PRs to just kinetic-devel and we cherry-pick them forward, but by default we'd like to move new development to the latest branch per best practices.

rhaschke, 2018-05-31:
I thought of this before and actually I would prefer to keep kinetic-devel as the default for now.
Most/all? maintainers still work on Kinetic. I suggest to wait until the melodic-branch can be build on Kinetic, i.e. until #925 is resolved.

v4hn, 2018-06-03:
I strongly agree with Robert.

As long as all maintainers who actually run robots use kinetic-devel to test, let's keep it the default.

bmagyar, 2018-06-03:
It's certainly convenient for daily work but it is also an implicit directive as to which development branch new PR-s should be opened against. With one you'll cherry-pick changes into the older distro, with the other they go to the newer one. It is only a matter of preference (once `melodic-devel` builds on Kinetic of course).

v4hn, 2018-06-04:
Quoting from [our website](http://moveit.ros.org/documentation/contributing/pullrequests/):

    Your pull-requests can target any branch of a MoveIt repository that supports an OSRF-supported ROS distribution. The most important point about your request is that you verified it is working. Obviously, the best way to do that is with the setup/distribution you use.

davetcoleman, 2018-10-18:
I'd like to switch to the melodic branch in time for World MoveIt! Day - this seems like a good time to have everyone help us test the new LTS and target changes there. We shouldn't be breaking kinetic at this point. Thanks to @IanTheEngineer we're very close to this: https://github.com/ros-planning/moveit/issues/925

I've just forked the tutorials and updated the website:
- https://github.com/ros-planning/moveit.ros.org/pull/205
- https://github.com/ros-planning/moveit_tutorials/pull/233

davetcoleman, 2018-10-24:
All MoveIt! repos have been updated to use Melodic branch by default, except where no Melodic branch existed.

closed 2018-10-24 by davetcoleman