moveit/moveit #930: Double visualization of robot motion group while operating through c++ Nodes.

Danfoa, 2018-05-31:
### Description
We have a file that will operate a simulated or real robot, through `MoveGroupInterface` class on a c++ Node, when simulating after launching the `demo.launch` (autogenerated by the setup assistant to our robot) the movements defined on the c++ file seem to perform correctly but Rviz representation of our robot seems to be duplicated; for example if the default planning group when launching the demo.launch is the left arm, through the hole operation this move group wont change position unless given through the GUI, and if the c++ file operates this move group a duplication of the arm visualization occurs (see below).

[image]
As you can see below the arm initial position is inside the drum and the last one is on top of the drum, but during all motions the orange arm is fixed there.

If we change the default move group at launching to be the right arm or the torso the same thing happens to that default move group during operation. This seems to be a miss configuration from our side but I am not sure what is wrong.

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* Source  build

### Steps to reproduce

Our repository is this [Danfoa/invite-robotics](https://github.com/Danfoa/invite-robotics), the demo.launch file is inside the moveit_config package and the c++ file we are testing can find  [here](https://github.com/Danfoa/invite-robotics/blob/developmental/invite_beginner_tutorials/src/invite_pick_and_place.cpp)

### Expected behaviour
When we execute motion through the MoveGroupInterface and a c++ Node on a move group, this move group should reflect the motion on Rviz even if it was selected as the "current planning group" on the GUI.

v4hn, 2018-06-03:
I'm not sure I understand your problem. It might help to add some explanations and arrows in the picture...

If you think the orange group in the display is redundant, just disable the "Query Goal State" property in the "Planning Request" subgroup. The orange visualization is the Interface to define a robot state in the RViz display, it is not relevant for anything else.

If you have a different problem, please elaborate.

rhaschke, 2018-07-04:
I think, its obvious that the orange robot visualization, originating from the `Query Goal State` property in rviz, should be disabled to solve this "issue".

closed 2018-07-04 by rhaschke