moveit/moveit #941: Collision Objects dissappear shortly after being added via PlanningSceneMonitor

partlygloudy, 2018-06-07:
### Description

I'm attempting to add a collision object to a PlanningSceneMonitor's monitored scene. I tried using the approach below which sends a moveit_msgs::collision_object message over the 'collision_objects' topic. It should then be received by the WorldGeometryMonitor and added to the planning scene. I have another node which listens to the planning scenes published by the monitor (over the "current_planning_scene" topic. Ultimately I want to check for collisions with that node but for now I am just trying to get the collision objects added properly. Currently, this node just counts the number of collision objects present in the received planning scene and prints the count. 

When I send the collision object once, as below, the 'receiver' node gets an updated planning scene containing the new collision object (as expected) and prints 1. No more planning scenes are sent out by the ps monitor until I move the robot, causing an update via the "joint_states" topic. When the second planning scene is received, the count is now 0 and there are no collision objects in the scene. So sometime between the initial message and the second message, the collision object has disappeared. Furthermore, the ps monitor never sends a message indicating that the collision object is no longer in the world as I would expect it to do. It's as though the collision object just disappears without any indication. 

The object seems to disappear after about 0.5 seconds. If I keep publishing collision objects with changing id's at 10 hz, the receiver node indicates that 5 objects are present at a given time. If I publish at 100 hz, ~50 are present. So it seems the collision objects are added, persist for half a second, and then disappear without the ps monitor indicating that they have disappeared. 

In addition to the approach below, I have tried sending the objects via planning scene diffs over the "planning_scenes" topic (using startSceneMonitor rather than startWorldGeometryMonitor). I also tried sending them using a PlanningSceneInterface's attachCollisionObjects method ( which I believe just uses the "planning_scenes" topic as well. Lastly, I tried to use the moveit_visual_tools 'processCollisionObjectMsg' method which bypasses ros and uses a LockedPlanningSceneRW to add the collision object. Exact same result in each case. 

One last thing... I receive a '[WARN] Returning dirty link transforms' whenever the collision object is published.

_Publisher_

```
int main(int argc, char **argv)
{
    // set up planning scene monitor
    planning_scene_monitor::PlanningSceneMonitorPtr psm_ptr = std::make_shared<planning_scene_monitor::PlanningSceneMonitor>("robot_description");
    psm_ptr->startStateMonitor("joint_states");
    psm_ptr->startWorldGeometryMonitor("collision_objects", "world", false);
    psm_ptr->startPublishingPlanningScene(psm_ptr->UPDATE_SCENE, "current_planning_scene");

    // collision obj publisher for adding objects to scene
    ros::Publisher collision_obj_publisher = n.advertise<moveit_msgs::CollisionObject>("collision_objects", 1);

    // Create table
    moveit_msgs::CollisionObject box;

    box.header.frame_id = "/world";
    box.id = "box"

    shape_msgs::SolidPrimitive primitive;
    primitive.type = primitive.BOX;
    primitive.dimensions.resize(3);
    primitive.dimensions[0] = 0.3;
    primitive.dimensions[1] = 0.3;
    primitive.dimensions[2] = 0.3;

    geometry_msgs::Pose box_pose;
    box_pose.orientation.w = 1.0;
    box_pose.position.x =  0.4;
    box_pose.position.y =  0.419;
    box_pose.position.z =  0.0;

    box.primitives.push_back(primitive);
    box.primitive_poses.push_back(box_pose);
    box.operation = box.ADD;

    // Wait for monitor to connect, publish collision obj
    ros::WallDuration sleep_t(0.5);
    while (collision_obj_publisher.getNumSubscribers() < 1)
    {
      sleep_t.sleep();
    }

    collision_obj_publisher.publish(box);

    // Loop
    ros::Rate loop_rate(10);
    while (ros::ok())
    {
        loop_rate.sleep();
        ros::spinOnce();
    }
    return 0;
}
```

_Subscriber_

```
planning_scene::PlanningScene *scene;    // initialized in main

void checkCollisions(const moveit_msgs::PlanningSceneConstPtr &msg)
{
    // Convert planning scene message to real planning scene
    scene->setPlanningSceneMsg(*msg);

    // DEBUG -----
    std::vector<std::string> ids = scene->getWorld()->getObjectIds();
    ROS_INFO_STREAM(ids.size());

}
```

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04

partlygloudy, 2018-06-08:
**UPDATE:**
I discovered that the issues are likely all due to the fact that while the first planning scene message sent by the monitor is not a diff, the subsequent messages sent are specified as diffs. This would lead to the problems outlined above as I was expecting those messages to be the complete scene rather than diffs. So even though the monitored planning scene is maintaining the collision objects I add, only the new ones are being sent, along with the new joint states. 

Correct me if I'm wrong here but my understanding was that if I set the SceneUpdateType to UPDATE_SCENE in the constructor for the monitor (as I did above), every planning_scene message sent out should be a full scene and none should really ever be diffs. 

If my understanding is correct, this could in fact be a bug, possibly caused by line 455 in planning_scene_monitor.cpp

`new_scene_update_ = (SceneUpdateType)((int)new_scene_update_ | (int)update_type);`

In the scenePublishingThread() method which is responsible for actually publishing the new planning_scene message whenever an update occurs, nothing happens until new_scene_update_ != UPDATE_NONE (which = 0). Then after the update is published, new_scene_update_ is reset to UPDATE_NONE. So in line 455, the left side of the logical OR should always be 0 and therefore isn't really doing anything. It seems possible that the line should have been 

`new_scene_update_ = (SceneUpdateType)((int)publish_update_types_ | (int)update_type);`

publish_update_types_ is set from the SceneUpdateType parameter passed in the constructor so this would ensure that new_scene_update would _at the very least_ be whatever 'level' of update specified in the constructor, due to the OR-ing. This seems to be the intended outcome of that line. So if we set the SceneUpdateType to UPDATE_SCENE, any call to triggerSceneUpdateEvent would result in the full scene being published, as expected.

I've been stuck on this issue for way too long so I totally understand if there's something obvious I've missed, but so far this is the best explanation I can come up with.

v4hn, 2018-06-08:
> scene->setPlanningSceneMsg(*msg);

This should be `scene->usePlanningSceneMsg(*msg);` to account for the diffs

v4hn, 2018-06-08:
> ```new_scene_update_ = (SceneUpdateType)((int)new_scene_update_ | (int)update_type);```

If you think you found a bug here, please provide a pull-request and explain the reasoning there, referencing surrounding code and examples there.

partlygloudy, 2018-06-08:
Yes but it seems like the triggerSceneUpdateEvent method is the only place where new_scene_update can be set to anything other than UPDATE_NONE. And as soon as it is set to anything else, the scenePublishingThread advances, publishes the appropriate message, and then resets it to UPDATE_NONE again. So when this line is reached the left side of the OR is always 0 so the line is effectively new_scene_update = update_type.

Just as a brief example 
-> we create the monitor passing UPDATE_SCENE (or 1111) as the SceneUpdateType
-> so publish_update_types_ = 1111
-> Now say there is an update to the joint states
-> triggerSceneUpdateEvent is called with UPDATE_JOINTS (or 0100) as update_type
-> at line 455 new_scene_update_ now = 0100
-> The loop at 349 and the if at 353 pass and we end up getting a diff message at 362

But we should never get a diff message as this contradicts the documentation for startPublishingPlanningScene: _"The first message sent out is a complete planning scene. Diffs are sent afterwards on updates specified by the event bitmask. **For UPDATE_SCENE, the full scene is always sent.**"_ 

Sorry I wasn't convinced there was a bug here when I first posted. If my logic here seems okay I will do a pull-request when I have a chance. In the meantime it should be simple enough to find a workaround now that I know what the real problem is.