moveit/moveit #943: Regression: DepthImageOctomapUpdater was removed from debs

mintar, 2018-06-08:
### Description

The depth image octomap updater stopped working in the newest released moveit version. The error message is:

```
[ERROR] [1528466140.174800759]: Exception while loading octomap updater 'occupancy_map_monitor/DepthImageOctomapUpdater': Could not find library corresponding to plugin occupancy_map_monitor/DepthImageOctomapUpdater. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
```

This is a regression in moveit_ros_perception 0.9.12 (released yesterday, 2018-06-07). The files `libmoveit_depth_image_octomap_updater.so` etc. are not contained in the released debs any more.
Here is the diff:

```
#### 0.9.11

/opt/ros/kinetic/lib/libmoveit_depth_image_octomap_updater.so
/opt/ros/kinetic/lib/libmoveit_depth_image_octomap_updater.so.0.9.11
/opt/ros/kinetic/lib/libmoveit_depth_image_octomap_updater_core.so
/opt/ros/kinetic/lib/libmoveit_depth_image_octomap_updater_core.so.0.9.11
/opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater.so
/opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater.so.0.9.11
/opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater_core.so
/opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater_core.so.0.9.11

#### 0.9.12

/opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater.so
/opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater.so.0.9.12
/opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater_core.so
/opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater_core.so.0.9.12
```

After looking through the commit history, commit 8609bde776a9d86a12f67ea16e750ab2b54339a6 (from PR #698) seems to be the one that broke the build.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04, normal x86_64 laptop
* Source or Binary build? binary
* If binary, which release version? 0.9.12

davetcoleman, 2018-06-20:
@bulwahn do you understand why your #698 has caused this? Shouldn't ``WITH_OPENGL`` default to ON when being released on the ROS buildfarms?

bulwahn, 2018-06-20:
@davetcoleman I just woke up and I will look at this today. I am surprised about the change in the release: it seems that the buildfarms must have explicitly set WITH_OPENGL to OFF (because of their own machine configuration) or there is something even more strangely going on.

davetcoleman, 2018-06-20:
@wjwwood could probably speak to whether WITH_OPENGL is set to OFF on the ros build farm...

mintar, 2018-06-20:
You can check for yourself: http://build.ros.org/job/Kbin_uX64__moveit_ros_perception__ubuntu_xenial_amd64__binary/76/consoleFull

It doesn't say anything about WITH_OPENGL. I guess you should be able to reproduce this by building everything from scratch in a new workspace.

mintar, 2018-06-20:
Found it! The following lines should have "WITH_OPENGL", not "BUILD_OPENGL":

https://github.com/ros-planning/moveit/blob/a989aa4b1f72b4ce939945c585d2f159d4f98b0e/moveit_ros/perception/CMakeLists.txt#L103-L106

mintar, 2018-06-20:
I've added a PR for the typo. Haven't tested it though.

cc143, 2018-06-21:
@mintar Your PR looks like it will fix #953 too... but it is failing the Travis CI checks. How does the current version of moveit have a [whole page](https://github.com/ros-planning/moveit) of green "passing" badges when it doesn't work but your PR fails the CI checks? There's something weird about the CI setup for moveit.

gavanderhoorn, 2018-06-21:
@cc143: the travis build failing does not seem related to the changes introduced by @mintar.

All the badges in the readme point to ROS buildfarm devel jobs, not to travis builds. So until PRs with failing Travis builds get merged into the respective `-devel` branches in `ros-planning/moveit`, the badges will stay green.

bulwahn, 2018-06-21:
Yes, I seemed to totally messed up that commit in my renaming of the option from BUILD_OPENGL to WITH_OPENGL when I did this in the last change before the merge.

cc143, 2018-06-21:
> As the Travis builds run for each PR, and the buildfarm jobs only upon merges, desync between these two states is possible.

OK, that makes sense. But if the Travis build for a PR is failing then it can only be for one of two reasons: either that PR has introduced a bug, or a bug was already merged. In the latter case, the buildfarm job should have failed after the bug was merged.

> It could be that the Travis tests are failing because of changes in dependencies that are not under our control

But unless a PR changes which dependencies you use, that shouldn't make any difference. You always build against the same versions of the same dependencies unless a PR changes that, no?

> Afaik the tests were not failing when the last release was done.

This is a bit surprising. The octomapper was missing completely so anything that uses it should have failed a test. Are there no tests of anything that uses octomapping?

gavanderhoorn, 2018-06-21:
This is getting slightly off-topic, but:

> But if the Travis build for a PR is failing then it can only be for one of two reasons: either that PR has introduced a bug, or a bug was already merged.

There is at least one other possible cause: a package that provides a dependency has recently been changed in an incompatible way, and the dependent has not been updated to take that change into account.

> But unless a PR changes which dependencies you use, that shouldn't make any difference.

No, see above.

> This is a bit surprising. The octomapper was missing completely so anything that uses it should have failed a test. Are there no tests of anything that uses octomapping?

While it's technically possible to write rostests that would exercise these parts of the infrastructure, it's not trivial. I may be misremembering, but there are currently no integration tests that touch the `octomap` I believe.

cc143, 2018-06-21:
It is getting off-topic, sorry. I'm just trying to understand how this could happen.

> a package that provides a dependency has recently been changed in an incompatible way, and the dependent has not been updated to take that change into account.

This is something that worries me. You seem to be saying that you just always build against the latest version of a dependency, with no version management. Is that really true?

I would normally expect a project to build against specific versions of dependencies. If a dependency changes in an incompatible way it doesn't matter because you are not using the new version, you are still using the old version before the incompatible change was made. The change to a newer version would then be made explicitly, in a pull request that also includes updates to take the dependency's changes into account.

>  I may be misremembering, but there are currently no integration tests that touch the `octomap` I believe.

OK, that makes sense then. Does ROS have any concept of stable releases? For example, I picked "ROS Kinetic Kame" as it was the latest "Long Term Support" release at the time. But LTS doesn't seem to equate to stable: breaking changes like this one are released, overwriting all the packages in the ROS apt repository. Did I pick the wrong version? Is there anything like Debian's _stable_ version?

mintar, 2018-06-21:
The way it works is like this: Whenever a maintainer does a "bloom release", the source gets tagged with a new version number and the build farm builds the package and all its dependents (and puts them into the ros-shadow deb repo). This can cause dependents to fail on the build farm. Every two weeks there's a sync, where the latest successful debs are copied from ros-shadow to the official ros deb repo.

So there is a two-step process that catches compilation errors, and guarantees that if a dependency breaks other packages it's not released. However, bugs like this one that are neither compilation errors, are not caught by a test, and are not manually caught by someone testing the ros-shadow repo make it into the "stable" ros repo.

It's a shame that the deb repo only keeps the latest version of a package and deletes all previous versions. It wouldn't be trivial to go back to a previous version anyway, because all packages are version-locked (so you would have to roll back *all* ROS packages to the previous version).

cc143, 2018-06-21:
> So there is a two-step process that catches compilation errors, and guarantees that if a dependency breaks other packages it's not released.

That's good, thank you for explaining. It still seems bad that the release process can cause unrelated CI failures on a pull request though: doesn't that make it hard to automatically gate PRs?

> It wouldn't be trivial to go back to a previous version anyway, because all packages are version-locked (so you would have to roll back _all_ ROS packages to the previous version).

In my case, the problem is really that I'm forced to go forward: I was using `apt` to install a particular package version but that failed as soon as a new release was made, since the version I wanted was no longer available. So I'm forced to always take the latest version whenever I build a new image. That wouldn't be so bad if there were a _stable_ branch, i.e. one that only received critical bugfixes, but I can't find such a thing for ROS.

Now that I know this is how ROS works, I will take a local snapshot of the repo once this bug is fixed and be careful not to update it. It's not the end of the world, just a bit awkward. Thanks for your help.

gavanderhoorn, 2018-06-21:
off-topic again, but:

> It wouldn't be trivial to go back to a previous version anyway, because all packages are version-locked (so you would have to roll back all ROS packages to the previous version).

Witihin ROSIN we're performing a study of historical ROS bugs and I've written a small script that actually allows you to generate a `.rosinstall` file for a point "in the past". You can then proceed with a source build of the ROS components for a specific set of pkgs that would allow you to do this (ignoring the fact that the state of the OS has progressed of course, but still).

It's not perfect, but works well enough for now: [rosinstall_generator_time_machine](https://github.com/rosin-project/rosinstall_generator_time_machine).

mintar, 2018-07-12:
Could we please merge this before the next release? It fixes a clear regression, and the longer it stays open, the more people will be affected.

closed 2018-07-12 by rhaschke

AndreaIncertiDelmonte, 2018-09-07:
@mintar the DepthImageOctomapUpdater is still not included into the _ros-kinetic-moveit-ros-perception                          0.9.12-1xenial-20180825-015504-0800_. Is the regression still in place? Would it be included in the next version of the package?

rhaschke, 2018-09-07:
We didn't have a new release yet. #954 fixes this and will be part of the next release.

AndreaIncertiDelmonte, 2018-09-07:
@rhaschke thanks :)

sjhansen3, 2018-10-08:
Is this fixed in the latest release @davetcoleman ?

davetcoleman, 2018-10-13:
@sjhansen3 no I'm afraid not, see https://github.com/ros-planning/moveit/issues/1083

PR #954 (2018-07-12): Clean up another typo in perception CMakeLists.txt
This fixes #943 .