moveit/moveit #944: New line in tutorial formatter

Ridhwanluthra, 2018-06-08:
### Description

Need to write the comments for the code in a new line. I am not sure how to do this using the tutorial-formatter

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source build

### Steps to reproduce
Just write a tutorial and type on different lines.

### Expected behavior
// hey
// hey1
should come as 
hey
hey1

### Actual behaviour
they come as
heyhey1

Ridhwanluthra, 2018-06-08:
@mlautman

v4hn, 2018-06-10:
See #945 for my feedback. The same applies here.
Also, if you have a particular problem with a tutorial you work on, please file a pull-request for it and add comments where you want feedback.

closed 2018-06-11 by Ridhwanluthra

mlautman, 2018-06-11:
Play around with /**/ to see if that gets you what you are looking for. If not, just keep trying different things and building locally until you find what you are looking for.