moveit/moveit #948: Using MoveGroupCommander.pick results in 'No Sampler was Constructed'

christgau, 2018-06-12:
### Description

I am using MoveIt's Python interface to control a Franka Emika Panda robot. I want to the pick method to grasp an object, that has been added to the scene. I can use `set_pose_target`, `plan` as well as `go` to move the arm into a certain pose. However, the `pick` method does not work at all. I constructed a scene and objects were added to it. I tried both the `pick` method without additional grasps and using a list of grasps as well.

In the result, the error message "No sampler was constructed" is shown. It is unclear to me, what's the root cause for this problem. I am primarily developing using `panda_moveit_config/demo_launch` but I also tried it with a real robot. In both cases, the error message is the same. Is there a missing parameter in the configuration/launch files? 

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build, 0.9.11-0xenial-20180514-063510-0800

### Steps to reproduce
Use the Python source code provided in the [Gist](https://gist.github.com/christgau/03119d7905c79f4501f07794e22fa8da) that also includes log output. The code provides no grasps, but even with some calculated grasps, the effect is the same.

### Expected behaviour
Pick executes and the robots grasps the object if possible.

### Actual behaviour
The error message "No sampler was constructed" is emitted. 

### Console output

Can be found in the [Gist](https://gist.github.com/christgau/03119d7905c79f4501f07794e22fa8da)

Ridhwanluthra, 2018-06-13:
I am not entirely certain that calling pick without a grasp message would work. I know its there but I am not sure how that works so will wait for someone else to comment. 

I am currently working on building a tutorial for this pick pipeline where you specify the grasp message and call pick with the object name and that message.

You can find a dirty but working version of this in c++ [here](https://github.com/Ridhwanluthra/moveit_tutorials/blob/pick_plan_tutorial/doc/pick_place/src/new_pick.cpp).

Feel free to ask anything

christgau, 2018-06-13:
> I am not entirely certain that calling pick without a grasp message would work. I know its there but I am not sure how that works so will wait for someone else to comment.

Well, as I wrote in my original message, I also tried calling pick with graps but it resulted in the same error message. My impression/assumption is that there is a fundamental problem with the robot configuration or with something else inside MoveIt. It appears to me that the sampler construction, which seems to fail, is something that I can't control from the application side.

> You can find a dirty but working version of this in c++ here.

Thanks a lot for the pointer! I'll try it in my development environment and provide feedback as soon as I can run the code.

christgau, 2018-06-14:
I compiled and executed the new C++ tutorial code within a simulation environment (`roslaunch panda_moveit_config demo.launch`) with updated and thus recent Kinetic MoveIt (ros-kinetic-moveit/xenial 0.9.12-1xenial-20180530-071129-0800) packages. Different to the observation of @Ridhwanluthra, the code does not work as expected, which would be the robot picks up the small cuboid ("cylinder)" that stands on a big box. What I get are error messages...

With the provided code unchanged I get the following error message from moveit

```
[ INFO] [1528967074.316118766]: Planning attempt 1 of at most 1
[ERROR] [1528967074.316692685]: No end-effector specified for pick action
[ INFO] [1528967075.410678068]: Planning attempt 1 of at most 1
[ERROR] [1528967075.416792564]: There are no end-effectors specified for group 'panda_arm'
```

new_pick fails with the following messages:

```
[ INFO] [1528967071.926976238]: Loading robot model 'panda'...
[ INFO] [1528967071.927040815]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1528967072.069155386]: Loading robot model 'panda'...
[ INFO] [1528967072.069198019]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1528967073.291160464]: Ready to take commands for planning group panda_arm.
[ WARN] [1528967074.317343113]: Fail: ABORTED: Must specify group in motion plan request
[ WARN] [1528967075.417489713]: Fail: ABORTED: Must specify group in motion plan request
```

The reason seems to be the wrong planning group ("panda_arm"). If I change it to "panda_arm_hand" I get the same error message as in my original example "no sampler was constructed":

(new_pick)
```
[ INFO] [1528968063.870933470]: Loading robot model 'panda'...
[ INFO] [1528968063.870996012]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1528968063.971941942]: Loading robot model 'panda'...
[ INFO] [1528968063.971997628]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1528968064.921635747]: Ready to take commands for planning group panda_arm_hand.
[ WARN] [1528968066.032732605]: Fail: ABORTED: No motion plan found. No execution attempted.
[ WARN] [1528968067.111935611]: Fail: ABORTED: Must specify group in motion plan request
```

(moveit)
```
[ INFO] [1528968066.014631837]: Planning attempt 1 of at most 1
[ INFO] [1528968066.027028825]: Added plan for pipeline 'pick'. Queue is now of size 1
[ERROR] [1528968066.027832028]: No sampler was constructed
[ INFO] [1528968066.027898740]: Manipulation plan 0 failed at stage 'reachable & valid pose filter' on thread 3
[ WARN] [1528968066.028244300]: All supplied grasps failed. Retrying last grasp in verbose mode.
[ INFO] [1528968066.028471887]: Re-added last failed plan for pipeline 'pick'. Queue is now of size 1
[ INFO] [1528968066.028709495]: Manipulation plan 0 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1528968066.028949940]: Pickup planning completed after 0.011968 seconds
[ INFO] [1528968067.110869299]: Planning attempt 1 of at most 1
[ERROR] [1528968067.110976762]: Attached body 'cylinder' not found

```

If I use the planning group `hand` the symptoms are identical to the `panda_arm` scenario: "No end-effector specified for pick action", which not surprising since the hand has no EEF. 

Does anyone have an idea, what's going on here and how to successfully use the pick and place routines within simulation (and a real Panda robot)?

Ridhwanluthra, 2018-06-15:
After your message I tried to run it again on my system and its works fine.

Could you please verify that you are using [this panda_moveit_config](https://github.com/ros-planning/panda_moveit_config) on master.

Also the planning group is panda_arm check [here](https://github.com/ros-planning/panda_moveit_config/blob/master/config/panda.srdf#L12)

If this is all ok, you can try with moveit source install but I dont think there is any new changes in this.

christgau, 2018-06-15:
Thanks for cross-checking. I followed your suggestion and checked the version/state of panda_moveit_config. On my system, ros-kinetic-panda-moveit-config/xenial,now 0.4.0-0xenial-20180530-070522-0800 is installed. While it is the latest package available from the ros distro packages and its date looks like it is from end of May, the package is in fact outdated compared to the github master. The latter's code is unchanged since about mid of May. In contrast, the code-base that was used to build the ros-package seems to be from end of March. 

Anyways, the outdated version appears to be root cause for the 'No sampler was constructed' error message. Using the ros-planning/panda_moveit_config@8bfd18ee6ee2f683aa7d479bd6a1beb9f6a9226e makes your code work and my code (with a changed planning group) does not throw the error message anymore. 

Thanks for your support.

closed 2018-06-15 by christgau