moveit/moveit #950: A way to use tf_prefix with virtual joint?

dnrohr, 2018-06-13:
### Description

I am trying to simulate two mobile robots, each with its own arm which I am trying to control with MoveIt. To organize the two robots, I am using namespacing on the nodes and topics and using tf_prefix to delineate to separate TF trees. To connect each robot to the world, I would like to use a planar virtual joint from base_link to odom. However, when tf_prefix is applied outside the urdf, these frames become /h1_tf/base_link and /h1_tf/odom. As a result, the TF lookup fails to find the frames with:

> [ WARN] [1528829858.191182394, 1630.340000000]: Unable to update multi-DOF joint 'virtual_joint': TF has no common time between '/odom' and 'base_link': 

The corresponding section of the code is in "/planning_scene_monitor/src/current_state_monitor.cpp":

```
      const moveit::core::JointModel* joint = multi_dof_joints[i];
      const std::string& child_frame = joint->getChildLinkModel()->getName();
      const std::string& parent_frame =
          joint->getParentLinkModel() ? joint->getParentLinkModel()->getName() : robot_model_->getModelFrame();

      ros::Time latest_common_time;
      std::string err;
      if (tf_->getLatestCommonTime(parent_frame, child_frame, latest_common_time, &err) != tf::NO_ERROR)
      {
        ROS_WARN_STREAM_ONCE("Unable to update multi-DOF joint '"
                             << joint->getName() << "': TF has no common time between '" << parent_frame.c_str()
                             << "' and '" << child_frame.c_str() << "': " << err);
        continue;
      }
```

I've tried hacking the srdf, but that seems to mess up things upstream. Is there a way to get this functionality to work, or am I going to have to modify the code for my application?

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build
* 0.9.12-1xenial-20180530-071129-0800

v4hn, 2018-06-14:
I don't think anyone here tried this before, so I would not expect this to work out of the box.

However, this seems to be a perfectly valid use-case that should be supported.
You will have to patch the code you cite. It would be nice if you could file a pull-request
for the change though and we would probably merge it soon.

dnrohr, 2018-06-14:
I have a hack in place that seems to work, and an idea to make it cleaner. I'd be happy to do a pull-request when I'm done.

Does it matter that 'tf_prefix' is deprecated? I don't like using it, but there doesn't seem to be a better alternative to separating the TF trees of multiple instances of a single robot type.

v4hn, 2018-06-18:
> I'd be happy to do a pull-request when I'm done.

Please do so!

> Does it matter that 'tf_prefix' is deprecated?

Personally, I don't care and fully agree with you.
The feature might not work correctly throughout all ROS packages (apparently including moveit), but apparently they decided to "deprecate" it in favor of ... nothing.
That simply does not make sense.

For reference: http://wiki.ros.org/tf2/Migration