moveit/moveit #962: AABB calculation for octree/octomap slightly off

simonschmeisser, 2018-06-26:
### Description

When profiling I noticed that we spend a lot of time with collision checks between the octree/octomap/occupancy_map and various meshes (both robot and attached objects). Looking at the calculation of the AABB for broad phase collision checks I noticed that this is very rough.

* AABB is assumed to be centered around some frame (camera? origin/world_frame in our case)
* size is calculated based on resolution and size. https://github.com/flexible-collision-library/fcl/blob/master/include/fcl/geometry/octree/octree-inl.h

In our case we have the following values:

tree_depth = 16
resolution = 0.001 (ie 1mm)

delta = (1 << 16) * 0.001 /2.0 = 32.768m

so we have a bounding box of length 65m for a robot cell smaller than one meter. I haven't found yet who sets the depth of 16 but in general I think we might want to calculate the AABB based on actual content.

OcTree does have some functions for calculating the size of the known world etc

This is a follow up on #488

v4hn, 2018-06-26:
> AABB is assumed to be centered around some frame (camera?

If the center is really the frame of the input point cloud, that would be pretty much useless.

> in general I think we might want to calculate the AABB based on actual content

Yes! Attempts to improve this are very welcome here.

simonschmeisser, 2018-06-27:
See https://github.com/flexible-collision-library/fcl/pull/309 for the PR. Currently the tests for octree all fail, I don't understand them unfortunately