moveit/moveit #963: New collision distance functionality gives nonsensical/inconsistent results

BoltzmannBrain, 2018-06-26:
### Description

The new collision distance functionality added in #662 (ping @rhaschke @davetcoleman) appears to be faulty (or hopefully we can just document it better and add some test cases as examples). Are there unit tests in moveit for this?

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build? Binary
* If binary, which release version? 0.9.12

### Example, behavior, etc.

Consider the following function:

```
std::map<std::string, double>
MyPlanningScene::CheckWorldCollisionDistance(const std::vector<double>& joint_angles) {

    moveit::core::RobotState query_state = GetRobotState(joint_angles);

    // Setup distance request and result data structs
    auto dreq = collision_detection::DistanceRequest();
    auto dres = collision_detection::DistanceResult();
    dreq.enable_signed_distance = false;
    dreq.type = collision_detection::DistanceRequestType::SINGLE;

    // Setup collision-models
    auto robot_model_ptr = planning_scene_->getRobotModel();  // `planning_scene_` is a pointer to a moveit::planning_scene::PlanningScene instance
    collision_detection::CollisionRobotFCL crobot(robot_model_ptr);
    auto world_ptr = planning_scene_->getWorldNonConst();
    collision_detection::CollisionWorldFCL cworld(world_ptr);

    // Compute obstacle distances for the given robot config
    cworld.distanceRobot(dreq, dres, crobot, query_state);

    std::cout << "min dist data...\n" <<
        "distance = " << dres.minimum_distance.distance << "\n"
        "link name = " << dres.minimum_distance.link_names[0] << "\n"
        "nearest pt = " << dres.minimum_distance.nearest_points[0] << "\n"
        "link name = " << dres.minimum_distance.link_names[1] << "\n"
        "nearest pt = " << dres.minimum_distance.nearest_points[1] << "\n"
        << std::endl;

    // Populate results dict
    // dres.distances is a ``moveit_core::collision_common::DistanceMap``
    std::map<std::string, double> d_per_link;
    for (const auto& p : dres.distances) {
        std::string link_name = p.first.first;
        double dist = p.second[0].distance;
        d_per_link[link_name] = dist;
    }
    return d_per_link;
}
```

After running (i.e. `std::out` calls not shown above) I print a results map {'link name' : world distance} for just the world objects with "link" in the name. I then run `planning_scene_->checkCollision` with a `CollisionRequest` and `CollisionResult` to check for consistent results.

For a robot configuration and scene known to not be in collision...

```
min dist data...
distance = -1
link name = work-table
nearest pt = 2.56933e-316
        -0.1
        0.55
link name = robot_cart
nearest pt = 0.00598614
 0.0317843
 0.0295336

DistanceResult ("links" only): {'base_link': 0.13404454375841973}
CollisionResult: {'distance': 0.01351107923187728, 'contact_count': 0, 'collisions': {}}
```

First printout implies a collision between 'work-table' and 'robot_cart' objects. The CollisionResult does not show a collision.

Again with `enable_signed_distance=true`, we actually get some distances for the robot links...

```
min dist data...
distance = -0.261899
link name = work-table
nearest pt = -0.0881485
   0.15378
 -0.388432
link name = upper_arm_link
nearest pt = -0.0881485
   0.15378
 -0.388432

{'base_link': 0.09794975379760043,
 'ee_link': 0.12001236570479465,
 'forearm_link': -0.05754458715578481,
 'hand_robotiq_85_base_link': 0.12957111350724262,
 'shoulder_link': 0.04981883428831674,
 'upper_arm_link': -0.261898698675521}
{'distance': 0.01351107923187728, 'contact_count': 0, 'collisions': {}}

```

Does the negative distance between 'work-table' and 'upper_arm_link' imply collision? CollisionResult doesn't think so. And what distance is CollisionResult showing?

Again with signed-distances, but also `dreq.type=collision_detection::DistanceRequestType::ALL`...

```
min dist data...
distance = -0.149511
link name = work-table
nearest pt = -0.220907
0.0993693
-0.293963
link name = forearm_link
nearest pt = -0.220907
0.0993693
-0.293963

{'base_link': 0.052649240629073835,
 'ee_link': 0.15425095967091407,
 'forearm_link': -0.14951081024320478,
 'hand_robotiq_85_base_link': 0.05336448225747473,
 'shoulder_link': 0.07109603742478011,
 'upper_arm_link': 0.051225287568354094}
{'distance': 0.01351107923187728, 'contact_count': 0, 'collisions': {}}
```

we have different distances across the board :frowning_face:

davetcoleman, 2018-06-27:
@BoltzmannBrain Why not ping original author (@Levi-Armstrong) of #662 ? 

 I'm afraid I do not know much about the collision checking internals so I hope others can assist here.

Also, it looks like you wrote up the start to some really good tests - can you turn these examples into a (failing) unit test so this issue does not occur again?

Levi-Armstrong, 2018-06-30:
It is possible to get inconsistent results when using the enable_signed_distance when objects are in collision. There is not an efficient way to get accurate penetration distances when two detailed meshes are in contact. It just looks at how far triangles penetrate each other so if you have small triangles it will report small penetrations when in-fact the objects could have large penetration distances. This is why it would nice to get new types added to URDF to indicate if a mesh is convex because there are algorithms to get accurate penetration depths between Convex-Convex and Concave-Convex shapes.

Although negative distances are currently expected to be inconsistent when objects are not in collision I would not expect it to be consistent. Is your current setup in a public repository that I can run and see if I can track down what is going on with the inconsistent positive distances?

BoltzmannBrain, 2018-06-30:
@Levi-Armstrong thank you for the info. It makes sense that repeat calls on a robot state that is in collision could give inconsistent negative distance values, but it seems that enabling the signed-distance flag changes the results completely. Or without that flag enabled are negative distances (i.e. objects in collision) ignored completely? Even more there's the issue that the traditional `CollisionResult` shows no collisions.

Unfortunately the code is not public, but in the above function `CheckWorldCollisionDistance` does the logic look correct for checking collision distances?

arocchi, 2018-08-21:
I wonder if this might be related to https://github.com/flexible-collision-library/fcl/issues/244
It would be interesting to try the test again, after many issues with EPA on libccd and fcl got fixed.