moveit/moveit #975: Tip frame for motion planning selectable

jschleicher, 2018-07-10:
[image]

For a rotated gripper we want to choose the tip-frame in addition to the planning group in RViz-MotionPlanning-plugin.

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* binary version ros-kinetic-moveit-ros-visualization 0.9.11-0xenial-20180423-181049-0800

### Steps to reproduce
Planning group up to flange, eef mounted as 
```
    <!-- instantiate the gripper -->
    <xacro:schunk_pg70 parent="$(arg robot_prefix)flange" name="$(arg gripper_name)" has_podest="false">
      <origin xyz="0 0 0.0171" rpy="0 0 ${-pi*3/4.}" />
    </xacro:schunk_pg70>
```

with a planning group in xacro 
```
    <end_effector name="pg+70" parent_link="${prefix}flange" group="gripper" parent_group="manipulator" />
```
### Expected behaviour
It should be possible to align the arrows with the gripper to move the end-effector in the direction of the gripper. Since the joint is fixed, this should be possible.

### Actual behaviour
The arrows are tilted 45 degrees (oriented with the flange), so it is hard to move the eef along its x-/y-axis.

felixvd, 2018-09-06:
I'm not sure what exactly you're requesting. Is this not possible already? For what it's worth, the way we align the interactive marker is to make the tip link of the robot the tip link of the gripper. From our moveit_config's SRDF:

    <group name="c_bot">
        <joint name="c_bot_shoulder_pan_joint" />
        <joint name="c_bot_shoulder_lift_joint" />
        <joint name="c_bot_elbow_joint" />
        <joint name="c_bot_wrist_1_joint" />
        <joint name="c_bot_wrist_2_joint" />
        <joint name="c_bot_wrist_3_joint" />
        <chain base_link="c_bot_base_link" tip_link="c_bot_robotiq_85_tip_link" />
    </group>
    <end_effector name="c_bot_gripper" parent_link="c_bot_robotiq_85_tip_link" group="c_bot" />

The tip_link is just a frame that we added manually for planning/carrying the interactive marker.

jschleicher, 2018-09-10:
So you would modify the robot SRDF for every gripper (should be possible) but furthermore modify the inverse kinemtics solver to calculate joint angles for this additional link. We have an optimized ikfast-based solver in the robot package that we do not suggest the user to re-generate.

I'd suggest to have a drop-down in the motion planning plugin panel (as for the planning group), where one can select the tip frame of the interactive marker (without modifying the SRDF).

I'd label this "enhancement". What do you think?

felixvd, 2018-09-11:
I think being able to set the interactive marker is a reasonable expectation, but I am not sure how well it works with the underlying implementation. I haven't looked at the code behind Rviz much.

v4hn, 2018-09-11:
There is code in place that will automatically account for fixed transforms between your IK solver tip and your tool point,
so you don't have to change your IK solver.

The drop-down menu sounds like a nice extension.
At the moment, the links for InteractiveMarker are decided behind a number of levels of abstraction
and are based on what you defined as `end_effector`, whether the group has a chain tip, ...
We would be happy to see an extension to that mechanism!

Please make sure you don't add any logic to the GUI code though that should actually be handled by `RobotState` et. al.