moveit/moveit #978: Procedure about to make a OMPL based path planner moveit plugin

lusu8892, 2018-07-10:
### Description

I want to make a OMPL based path planner moveit plugin, so that I can use the feature from OMPL and implement my own planning algorithm working as Moveit Plugin. I know this is possible, but I could not know how to do it. I saw someone on google group had the same issue.

Also, I want to know do I need to compile OMPL and Moveit from source if want to use OMPL library.

### Your environment
* ROS Distro: [IKinetic]
* OS Version: e.g. Ubuntu 16.04

atomoclast, 2018-07-31:
Very cool problem! I do find bimanual manipulation pretty cool. Something I would love to mess around with more in the future...as you mentioned, that's a higher level issue. Any literature I've seen treats it as such. So you should be able to wrap the current MoveIt API for each manipulator and plan to specific poses with required grasping forces and so on, which is derived from a task-based approach.

This paper may be helpful:
[Skill-Based Bimanual Manipulation Planning](http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.925.7193&rep=rep1&type=pdf)

DebasmitaGhose, 2018-07-31:
I have been trying to do a simplified version of this where I try to do a bimanual pick and place for an object at a known position of a known shape. I havent been able to find a reference code for that, so I have been trying to modify [this tutorial code](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pick_place/pick_place_tutorial.html) for my purpose. 

It would be great if you could provide some other resources that could help me perform this task.