moveit/moveit #985: Cannot load iris xacro in ros moveit

zubair123khan, 2018-07-14:
### Description

I have recently installed ROS moveit and i was trying to test IRIS quad in moveit, although when i am trying to upload iris.xacro from my px4 firmware, it showing error that "Undefined substitution argument rotors_description_dir None" and it not able to recognnize rotors description package in px4 firmware.
can anyone tell me i am forgetting to do any workaround !

Thanks 

### Your environment
* ROS Distro: [Kinetic]
* OS Version: 16.04

### Steps to reproduce
1. clone the px4 firmware from ### Description

I have recently installed ROS moveit and i was trying to test IRIS quad in moveit, although when i am trying to upload iris.xacro from my px4 firmware, it showing error that "Undefined substitution argument rotors_description_dir None" and it not able to recognnize rotors description package in px4 firmware.
can anyone tell me i am forgetting to do any workaround !

Thanks 

### Your environment
* ROS Distro: [Kinetic]
* OS Version: 16.04

### Steps to reproduce
1. clone the px4 firmware from https://github.com/PX4/Firmware
2. launch rosmove it 
3. load iris.xacro from rotor description

### Expected behaviour
i should see iris model loaded using .xacro

### Actual behaviour
ERROR : cannot find package rotor description 

### Backtrace or Console output
when processing file: /home/zubair/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro
[rospack] Error: package 'sitl_gazebo' not found
[librospack]: error while executing command
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
Undefined substitution argument rotors_description_dir None
when processing file: /home/zubair/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris.xacro

rhaschke, 2018-07-15:
You have several issues, all unreleated to MoveIt:
1. package sitl_gazebo is missing
2. You need to provide a command line argument `rotors_description_dir` when calling xacro.

I suggest to first figure out how to call xacro on the commandline.

zubair123khan, 2018-07-15:
Hello

Thanks for the reply , but it's not about command line, that's the output
of console when I tried to load iris.xacro in ros moveit window and hence I
got that error.

rhaschke, 2018-07-16:
I guess, when you are writing "load iris.xacro in ros moveit window", you actually mean to load this file in MoveIt's setup assistant, don't you?
If so, please try to get xacro running first on the command line like this:
`xacro --inorder iris.xacro`

The repository you mentioned, doesn't contain `iris.xacro`. Hence, I cannot reproduce your issue.

zubair123khan, 2018-07-16:
I will surely try your suggestion sir, also the repo is inside px4
firmware, it's in firmware/tools/sitl_gazebo/rotor_description/iris.xacro

rhaschke, 2018-07-16:
You should have mentioned that the repo uses `submodules`, which need to be cloned separately.
Looking into the urdf description, I find two substitution arguments:
`$(arg visual_material)` and `$(arg rotors_description_dir)`. These need to be passed as cmdline arguments to `xacro`, e.g. like this:
`xacro --inorder iris.xacro rotors_description_dir:=.. visual_material:=green`

These arguments can also provided in MoveIt's setup assistant.

zubair123khan, 2018-07-16:
I will surely implement this today,  i will close this if i dont see any
problem, thanks a lot for the help :)

zubair123khan, 2018-07-16:
Not able to run it :(.
To be clear i am pointing towards this xacros
https://github.com/PX4/sitl_gazebo/tree/ac54b8fdddd9b21dbe6d59deaf46494864858bc9/models/rotors_description/urdf

zubair123khan, 2018-07-16:
> gazebo/tree/ac54b8fdddd9b21dbe6d59deaf46494864858bc9/models/rotors_

This is what i am passing now.
Exactly what you told

zubair123khan, 2018-07-16:
xacro --inorder iris.xacro
rotors_description_dir:=/Home/Firmware/Tools/sitl_gazebo/models/rotors_description
visual_material:=green

zubair123khan, 2018-07-16:
ERROR in console :

xacro: error: expected exactly one input file as argument
[rospack] Error: package 'sitl_gazebo' not found
[librospack]: error while executing command
Usage: xacro [options] <input>

rhaschke, 2018-07-17:
Works for me, both in cmdline and MSA. Do you use a non-default source branch or do you modified some files?

zubair123khan, 2018-07-17:
Can you +please show your moveit screen (xacro arguments passed) !
No i did not modified anything :(

zubair123khan, 2018-07-17:
just want to see what arguments you are passing and i will check accordingly

rhaschke, 2018-07-17:
I provide exactly the options I state previously: `rotors_description_dir:=.. visual_material:=green`
[image]

To get rid of the remaining errors, you need to build your repo in a catkin workspace and source it.

zubair123khan, 2018-07-17:
> thats the problem, i am not able to build this firmware with a catkin
> package, i think its not supported. We need o build the environment outside
> of catkin.
> Also, the difference is that, i am taking the Firmware from my "home" and
> not from tmp like you did.
> Does this makes a difference ?
>
> thanks

Anyways i will surely try this again this evening, with my package in
catkin workspace.
I will then accordingly let you know :)
Thanks Robert

zubair123khan, 2018-07-17:
I have done it, but i am not able to see my drone on the right side while
the upload is done, i think its because of my environment not being catkin.
Also after making a package of my drone from setup assistance (which i
saved in a catkin build environment), i ran roslaunch iris_config
demo.launch and this is what i see.

.. logging to
/home/zubair/.ros/log/438c241a-89fb-11e8-9dbe-10f00557463a/roslaunch-zubair-HP-Pavilion-Notebook-5562.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line
306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py",
line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py",
line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py",
line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py",
line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
748, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename,
argv=argv, verbose=verbose)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
720, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context,
None, is_core, verbose)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
684, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine,
is_core, verbose)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
95, in call
    return f(*args, **kwds)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
626, in _include_tag
    default_machine, is_core, verbose)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
664, in _recurse_load
    self._param_tag(tag, context, ros_config, verbose=verbose)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
95, in call
    return f(*args, **kwds)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
256, in _param_tag
    vals = self.opt_attrs(tag, context, ('value', 'textfile', 'binfile',
'command'))
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
202, in opt_attrs
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line
183, in resolve_args
    return substitution_args.resolve_args(args,
context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py",
line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py",
line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py",
line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File
"/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py",
line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in
get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: sitl_gazebo
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/zubair/iris_ws/src
ROS path [2]=/home/zubair/catkin_ws/src
ROS path [3]=/opt/ros/kinetic/share

zubair123khan, 2018-07-17:
We cannot build px4 firmware with catkin (fatal error : ecl.h not found),
and it build and used normally outside catkin environment.

rhaschke, 2018-07-18:
Again, the error is that your sitl_gazebo package wasn't yet build as a catkin package and announced to ROS. However, building your px4 package is completely out of the scope of MoveIt. Please refer to this packages's maintainers for more help. I will close this issue for now.

closed 2018-07-18 by rhaschke