moveit/moveit #988: bad_alloc in collision_detection::distanceCallback

troizky, 2018-07-18:
### Description

CollisionGeometryData has union named `ptr`
```
union
  {
    const robot_model::LinkModel* link;
    const robot_state::AttachedBody* ab;
    const World::Object* obj;
    const void* raw;
  } ptr;
```
Getting `ptr.obj->id_` causes exception in case of AttachedBody in following lines:
https://github.com/ros-planning/moveit/blob/be78e68585f1552f2cca44d0eff3fbf87bbf358d/moveit_core/collision_detection_fcl/src/collision_common.cpp#L520
https://github.com/ros-planning/moveit/blob/be78e68585f1552f2cca44d0eff3fbf87bbf358d/moveit_core/collision_detection_fcl/src/collision_common.cpp#L521

### Environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* MoveIt Version: kinetic-devel
* Commit: https://github.com/ros-planning/moveit/commit/be78e68585f1552f2cca44d0eff3fbf87bbf358d

### Steps to reproduce
Exception throws when a robot tries to attach something.

### Backtrace or Console output
(gdb) where
#0  0x00007ffff2a31c7d in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff2a3222c in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00000000006051e8 in __gnu_cxx::new_allocator<char>::allocate (this=0x7fffca7f91e8, __n=140736280719801) at /usr/include/c++/7/ext/new_allocator.h:111
#3  0x00000000005fe86a in std::allocator_traits<std::allocator<char> >::allocate (__a=..., __n=140736280719801) at /usr/include/c++/7/bits/alloc_traits.h:436
#4  0x00000000006057a8 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create (this=0x7fffca7f91e8, __capacity=@0x7fffca7f8e88: 140736280719800, __old_capacity=15) at
 /usr/include/c++/7/bits/basic_string.tcc:153
#5  0x0000000000605271 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign (this=0x7fffca7f91e8, __str="") at /usr/include/c++/7/bits/basic_string.tcc:265
#6  0x00000000005fe8fd in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::assign (this=0x7fffca7f91e8, __str="") at /usr/include/c++/7/bits/basic_string.h:1347
#7  0x00000000005f9e70 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator= (this=0x7fffca7f91e8, __str="") at /usr/include/c++/7/bits/basic_string.h:685
#8  0x00007fffec8facc5 in collision_detection::distanceCallback (o1=0x1871ab0, o2=0x7fffb804bec0, data=0x7fffca7f9480, min_dist=@0x7fffca7f93c8: 1.7976931348623157e+308) at /root/ws_moveit/src/moveit/movei
t_core/collision_detection_fcl/src/collision_common.cpp:521
#9  0x00007fffe49a3a1b in fcl::details::dynamic_AABB_tree::distanceRecurse(fcl::NodeBase<fcl::AABB>*, fcl::CollisionObject*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*, double&), d
ouble&) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#10 0x00007fffe49a3a1b in fcl::details::dynamic_AABB_tree::distanceRecurse(fcl::NodeBase<fcl::AABB>*, fcl::CollisionObject*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*, double&), d
ouble&) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#11 0x00007fffe49a3a55 in fcl::details::dynamic_AABB_tree::distanceRecurse(fcl::NodeBase<fcl::AABB>*, fcl::CollisionObject*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*, double&), d
ouble&) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#12 0x00007fffe49a3a1b in fcl::details::dynamic_AABB_tree::distanceRecurse(fcl::NodeBase<fcl::AABB>*, fcl::CollisionObject*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*, double&), d
ouble&) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#13 0x00007fffe49a54a7 in fcl::DynamicAABBTreeCollisionManager::distance(fcl::CollisionObject*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*, double&)) const () from /usr/lib/x86_64-
linux-gnu/libfcl.so.0.5
#14 0x00007fffec931bf3 in collision_detection::CollisionWorldFCL::distanceRobot (this=0x7fffb84219d0, req=..., res=..., robot=..., state=...) at /root/ws_moveit/src/moveit/moveit_core/collision_detection_f
cl/src/collision_world_fcl.cpp:278
#15 0x00007ffff70d4e97 in collision_detection::CollisionWorld::distanceRobot (this=0x7fffb84219d0, robot=..., state=..., acm=..., verbose=false) at /root/ws_moveit/install/include/moveit/collision_detectio
n/collision_world.h:209
#16 0x00007ffff70d520e in planning_scene::PlanningScene::distanceToCollisionUnpadded (this=0x7fffb8647480, kstate=...) at /root/ws_moveit/install/include/moveit/planning_scene/planning_scene.h:630

troizky, 2018-07-18:
Method `CollisionGeometryData::getID()` should be used here.

rhaschke, 2018-08-20:
Fixed with #989. Thanks for reporting and providing the fix!

closed 2018-08-20 by rhaschke