moveit/moveit #991: connceting vrep to moveit

HelloDusty, 2018-07-20:
### Description
Hi, I am Mike.
I am doing connecting vrep to moveit by ros, coding hardware interface. The hardware interface can successfully read the robot jointstate into topic /joint_states in ros. It seemed like I have done the work, however, moveit can't fetch the robot current state and also can't execute the plan and report 
"Didn't received robot state (joint angles) with recent timestamp within 1 seconds. 
Check clock synchronization if your are running ROS across multiple machines!
[ WARN] [1532052771.859175643]: Failed to fetch current robot state."
Is there something wrong in my hardware interface? I launch UR URDF into ROS and load UR .ttt file into vrep. When I connecting ROS to real robot and vrep, real robot and vrep robot both publish robot states into topic /joint_states with some joints names. When moveit fetch robot state, moveit identify real robot  and only fetch real robot joint state in ros and filter vrep robot. How could moveit do it? 
My final question is how moveit identity robot and fetch its joint state. 

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

HelloDusty, 2018-07-20:
Here is my part of hardware interface 
`#include "MyRobot_vrepHW.h" 
#include <stdio.h>        
#include "../v_repLib.h"
#include <string>
#include <iostream>
#include <unistd.h>
using namespace std;

namespace ur
{

std::string MyRobot_vrepHW::sm_jointsName[MR_JOINTS_NUM] = {
    "shoulder_pan_joint",
    "shoulder_lift_joint",
    "elbow_joint",
    "wrist_1_joint",
    "wrist_2_joint",
    "wrist_3_joint"
};

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
MyRobot_vrepHW::MyRobot_vrepHW() :
    hardware_interface::RobotHW()
{	

	init();
    // Init arrays m_cmd[], m_pos[], m_vel[], m_eff[].
    for (int i = 0; i < MR_JOINTS_NUM; ++i)
    {
        m_cmd[i] = 0.0;
        m_pos[i] = 0.0;
        m_vel[i] = 0.0;
        m_eff[i] = 0.0;
        last_cmd[i] = 0.0;
    }

    // Init and get handles of the joints to control.
    for (int i = 0; i < MR_JOINTS_NUM; ++i)
        m_vrepJointsHandle[i] = -1;

    // Register joint interfaces.
    registerHardwareInterfaces();
}

	MyRobot_vrepHW::~MyRobot_vrepHW(){

	}
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
bool MyRobot_vrepHW::init()
{
    // Get joint handles.
    for (int i = 0; i < MR_JOINTS_NUM; ++i)
    {
        int vrepJointsHandle = simGetObjectHandle(sm_jointsName[i].c_str());

        if (vrepJointsHandle == -1)
        {
            ROS_ERROR_STREAM("ur robot interface not able to get handle for '" << sm_jointsName[i] << "'." << std::endl);

            return false;
        }

        m_vrepJointsHandle[i] = vrepJointsHandle;
    }

    return true;
}

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
void MyRobot_vrepHW::registerHardwareInterfaces()
{
    for (int i = 0; i < MR_JOINTS_NUM; ++i)
    {
        // Joint state interface.
        hardware_interface::JointStateHandle jointStateHandle(sm_jointsName[i], &m_pos[i], &m_vel[i], &m_eff[i]);
        m_jointState_interface.registerHandle(jointStateHandle);

        // Joint command interface (in MyRobot's case this is a velocity interface).
        //hardware_interface::JointHandle jointVelocityHandle(jointStateHandle, &m_cmd[i]);
        hardware_interface::JointHandle jointPositionHandle(jointStateHandle, &m_cmd[i]);
        //m_jointVelocity_interface.registerHandle(jointVelocityHandle);
        m_jointPosition_interface.registerHandle(jointPositionHandle);
    }

    registerInterface(&m_jointState_interface);
    //registerInterface(&m_jointVelocity_interface);
    registerInterface(&m_jointPosition_interface);
}

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
bool MyRobot_vrepHW::read()
{
    for (int i = 0; i < MR_JOINTS_NUM; ++i)
    {
        float pos,
              vel,
              eff;

        if (simGetJointPosition(m_vrepJointsHandle[i], &pos) == -1 ||
            simGetObjectFloatParameter(m_vrepJointsHandle[i], 2012, &vel) == -1 || // Velocity.
            simGetJointForce(m_vrepJointsHandle[i], &eff) == -1)
        {
            ROS_ERROR_STREAM("MR robot interface not able to get state for '" << sm_jointsName[i] << "'." << std::endl);

            return false;
        }
        m_pos[i] = pos;
        m_vel[i] = vel;
        m_eff[i] = eff;
    }
        m_pos[0] = m_pos[0] + 3.14;
        m_pos[1] = m_pos[1] - 1.57; 
        m_pos[3] = m_pos[3] - 1.57;
    return true;
}

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
bool MyRobot_vrepHW::write()
{
    for (int i = 0; i < MR_JOINTS_NUM; ++i)
    {   
        if(last_cmd[i] != m_cmd[i])
        {
        
            if(i == 0)
                m_cmd[i] = m_cmd[i] - 3.14;
            if(i == 1)
                m_cmd[i] = m_cmd[i] + 1.57;
            if(i == 3)
                m_cmd[i] = m_cmd[i] + 1.57; 
            //cout<<m_cmd[i]<<"//////";
            last_cmd[i] = m_cmd[i];
        }
        /*
        if(i == 0)
        m_cmd[i] = m_cmd[i] - 3.14;
        if(i == 1)
        m_cmd[i] = m_cmd[i] + 1.57;
        if(i == 3)
        m_cmd[i] = m_cmd[i] + 1.57;       */
        //cout<<m_cmd[i]<<"//////";
        //printf("/n")
        //if (simSetJointTargetVelocity(m_vrepJointsHandle[i], m_cmd[i]) == -1)
        if (simSetJointTargetPosition(m_vrepJointsHandle[i], m_cmd[i]) == -1)
        {
            ROS_ERROR_STREAM("MR robot interface not able to get state for '" << sm_jointsName[i] << "'." << std::endl);

            return false;
        }
    }

    return true;
}

} // namespace MR.
`

felixvd, 2018-09-06:
You should post this on [ROS Answers](https://answers.ros.org/) with some more troubleshooting info and clarifications. I don't really understand when your problem happens and when it doesn't, for example. This page is not the right place though, most likely.