moveit/moveit #1000: release indigo

v4hn, 2018-07-25:
We want to have another indigo release to distribute the last few bugfixes in the indigo-devel branch.

Unless users request other patches in indigo, this might very well be the last release for this ROS distribution.
We stopped cherry-picking back to indigo for quite a while now
and with some effort the `kinetic-devel` branch can be compiled and used on Ubuntu 14.04 too.

I will run the ABI checks and report back until the end of the week.

gavanderhoorn, 2018-07-25:
Nice, issue #1000.

v4hn, 2018-08-01:
@130s 

The only ABI-incompatible changes I found are https://github.com/ros-planning/moveit/pull/443
@davetcoleman removed the alternative time parameterization method again, because it was supposed to be replaced by the better alternative.
However, the new alternative was not merged into `indigo-devel` anymore.

I believe it is ok to release the branch as is. Alternatively, we could backport the newer spline parameterization method to provide another alternative for indigo.

davetcoleman, 2018-08-01:
Ah I hadn't realized the newer [time parameterization plugin](https://github.com/ros-planning/moveit/pull/441) was kinetic only. I think it makes sense to cherry-pick back to indigo, but it also probably doesn't matter much since its a new barely used feature that doesn't belong in an old LTS

130s, 2018-08-01:
Does releasing Kinetic and/or Melodic together make sense, or does that need a separate discussion?

v4hn, 2018-08-06:
> Does releasing Kinetic and/or Melodic together make sense, or does that need a separate discussion?

To be honest I don't know right now.

Given that GSoC will end soon I lean towards having the next release kinetic+ *after* the results get merged.

130s, 2018-08-15:
I see.

I'm planning to make release on this weekend, for Indigo only.

130s, 2018-08-30:
Can anyone list the ticket numbers that need to be addressed before Indigo release?

v4hn, 2018-08-31:
> Can anyone list the ticket numbers that need to be addressed before Indigo release?

None to the best of my knowledge.

130s, 2018-09-01:
Ah, then I might have misunderstood the few previous posts like https://github.com/ros-planning/moveit/issues/1000#issuecomment-413307231. So now my understanding: indigo release is ready (and it has been...turned out I've just slacked off from it. I thought there were blockers). I'll get to it this weekend.

rhaschke, 2018-09-01:
I think #1000 was mainly addressing Kinetic/Melodic blockers - despite the original issue title ;-)

davetcoleman, 2018-09-06:
Indeed, this issue was suppose to just be about releasing kinetic.

@130s let's get that released! :-)

130s, 2018-10-05:
Sorry to keep this pushed back. Will be done by WMD 2018.

davetcoleman, 2018-10-13:
friendly ping @130s

130s, 2018-10-21:
Please hold backport to Indigo as I'm planning to make a release into Indigo very soon (I doubt there are maintainers who do this though).

130s, 2018-10-21:
Done https://github.com/ros/rosdistro/pull/19320

130s, 2018-10-21:
https://github.com/ros/rosdistro/pull/19320 is closed due to the pre-release test failure https://github.com/ros-planning/moveit/issues/1106.

130s, 2018-10-23:
Steps to do to make this release happen:
- [ ] Resolve circular dependency -> PR https://github.com/ros-planning/moveit/pull/1111
- [ ] Update rosdistro indigo about the moveit_full_pr2 relocation (currently its [in moveit](https://github.com/ros/rosdistro/blob/d578284e09cfefefa780b6f06d7ae5767d33b58a/indigo/distribution.yaml#L7461) section).
  - While this can be done either by making release of moveit indigo, cleanest is to open a PR to rosdistro to just update the location and nothing else. Once that's done, nothing further shouldn't be neeed regarding moveit_full_pr2 location (no version update for moveit_full_pr2 so no new release shouldn't be needed, unless we want to weave in the information about relocation explicitly in the release info).
- [ ] Make Indigo release from moveit repo.

130s, 2018-10-25:
Done https://github.com/ros/rosdistro/pull/19347

closed 2018-10-25 by 130s