moveit/moveit #1006: Pick and Place Pipeline

DebasmitaGhose, 2018-07-30:
I was trying to use the pick and place pipeline but I dont understand how to start working with it. It would be great if somebody could explain how to get started with it.

atomoclast, 2018-07-31:
The MoveIt! has a great tutorial section.

There's a tutorial section for this pick and place tutorial already written:

[PIck and Place Tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pick_place/pick_place_tutorial.html#getting-started
)

DebasmitaGhose, 2018-07-31:
I have been using this tutorial, but it is not completely clear how the values of the grasp pose has been calculated.

felixvd, 2018-08-01:
The grasp pose has not been calculated, it was set up manually for that particular robot and object. There is no grasp candidate generation in MoveIt.

You also asked a very similar question in [this issue]. Please don't pollute the repositories with multiple  support requests and use ROS Answers instead. I also strongly suggest to spend more time troubleshooting by yourself and to read [this](https://stackoverflow.com/help/how-to-ask). "I am not able to figure out how this should be formatted" is a bad question because it does not show what you tried, and it is easily answered by reading the tutorial source code.

mlautman, 2018-08-02:
This is not an issue with the tutorials. Please ask questions like this on [ROS answers](http://answers.ros.org)

closed 2018-08-02 by mlautman