moveit/moveit #1009: Can't build because of a moveit_ros_perception error.

MohmadAyman, 2018-07-30:
### Description
After using `catkin clean`, `moveit_ros_perception` started giving an error when I run `catkin build` 

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source: latest

### Steps to reproduce
`catkin clean`
`catkin build`

### Backtrace or Console output
```
/usr/bin/ld: CMakeFiles/moveit_occupancy_map_server.dir/src/occupancy_map_server.cpp.o: undefined reference to symbol '_ZNK3ros9Publisher7publishERKN5boost8functionIFNS_17SerializedMessageEvEEERS3_'
/opt/ros/kinetic/lib/libroscpp.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [/home/mh/ws_moveit/devel/.private/moveit_ros_perception/lib/moveit_ros_perception/moveit_occupancy_map_server] Error 1
make[1]: *** [occupancy_map_monitor/CMakeFiles/moveit_occupancy_map_server.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

davetcoleman, 2018-07-30:
Looks like a build problem with how you are sourcing your catkin workspace, what version of debians you are using, etc...

v4hn, 2018-08-20:
closing as outdated.

closed 2018-08-20 by v4hn