moveit/moveit #1015: Obstacle information with out Depth camera !!

zubair123khan, 2018-08-03:
### Description

Hello 

I am developing obstacle avoidance using monocular vision and eventually working on obstacle relative size change. I am a node working which is subscribing to image_raw topic from cam and get the information and size of the obstacle in front. 
On top of this,  i am willing to use Moveit, so that i can command my drone to avoid accordinly. 
Currently I see that Moveit have just 2 pluggin Laser scanning and Depth camera mapping. 
I want to know what type of environment setting do i have to do so as to get the path planning with respect to obstacle information from my node. 

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

My intention is to use Moveit for path planning without having depth camera. 

Any suggestions !

Thanks

v4hn, 2018-08-20:
This should be a question on http://answers.ros.org

The simplest way for you to implement this is to just add collision objects to the planning scene yourself.
If you feel this is not "internal" enough for your application, e.g. in terms of update rate, you could also implement an alternative update plugin for the octomap.

closed 2018-08-20 by v4hn

zubair123khan, 2018-08-20:
Thanks for the answer, Although can you give me a sample reference on how
can i publish those obstacle coordinates in planning scene topic ! I would
so much glad.
Thanks

v4hn, 2018-08-20:
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/planning_scene_ros_api/planning_scene_ros_api_tutorial.html

zubair123khan, 2018-09-01:
To be added, i have read the document given by @v4hn and i have not seen in anywhere the documentation for how can i actually provide the coordinated of the object from my image processing node ? can you please help !!

felixvd, 2018-09-02:
> This should be a question on http://answers.ros.org