moveit/moveit #1021: move_group_interface_tutorials not showing expected trajectories for panda arm

shreyasgan, 2018-08-13:
### Description
When I run the move_group_interface_tutorials with the panda arm, and step through the various commands, and compare them to the video provided by the Moveit tutorials, the trajectories for setPoseTarget commands are not correct. There is a discrepancy between the desired and actual behavior. Also, the cartesian path planner is only able to plan 47.62% of the desired path. 

Is there something wrong with the way I set up Moveit? I don't really know how to fix this problem, so help would be much appreciated!
Thank you!

### Your environment
* ROS Distro: Kinetic
* OS Version:  Ubuntu 16.04
* Built from Moveit tutorials website
* sudo apt install ros-kinetic-moveit

### Steps to reproduce
1. run move_group_interface_tutorials with the panda arm demo.launch and step through the commands.

### Expected behaviour
XYZ desired position: [0.28, -0.2, 0.5]

### Actual behaviour
XYZ actual position: [0.3595, -2.09821e-12, 0.643499]
[ INFO] [1534019263.491940293]: Visualizing plan 4 (Cartesian path) (47.62% acheived)

### Backtrace or Console output

Ridhwanluthra, 2018-08-27:
could you test it again after changing your fake_controllers.yaml in config folder in panda_moveit_config to:
```
controller_list:
  - name: fake_panda_arm_controller
    joints:
      - panda_joint1
      - panda_joint2
      - panda_joint3
      - panda_joint4
      - panda_joint5
      - panda_joint6
      - panda_joint7
  - name: fake_hand_controller
    joints:
      - panda_finger_joint1
      - panda_finger_joint2

initial:
  - group: panda_arm
    pose:  home
```

mlautman, 2018-08-27:
@shreyasgan Thank you for bringing this up but in the future, please raise this issue in the moveit_tutorials repo for a faster response. You are correct that the video is out of date. Franka Emika recently changed the joint limits in franka_description causing the default configuration on launch to be hunched over rather than stood straight up. This affected the start state of the trajectories and is likely to be causing the issues that you describe. It would be great if you were able to take the time to fix this issue in the tutorials and create a new video

mlautman, 2018-08-27:
closing because this should live in moveit_tutorials

closed 2018-08-27 by mlautman