moveit/moveit #1026: Multiple-query planners in OMPL - maintain planner's information from the previous run

wolledav, 2018-08-20:
When using OMPL planners via MoveGroup interface, multiple-query planners such as PRM can't be used properly, as MoveIt! clears previous query data (more detailed description [here](https://answers.ros.org/question/300720/prm-roadmap-building-not-working-through-movegroup/).

I think that there should be the possibility to maintain planner data - current implementation probably intents to prevent collisions due to changes in PlanningScene, but it also makes it impossible to use multiple-query planners meaningfully.

* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

v4hn, 2018-08-20:
I agree with the whole thread on ros-answers.

It makes sense to add this capability - making the clear() optional, one way or another.

I also agree with Bryce in that you are very welcome to file a pull-request! :sunglasses: 

The obvious choice would be to clear the data only if some meaningful change in the PlanningScene occured between the queries. You might have to work your way through some layers of code to get that information.
Alternatively, if you are happy with changes for melodic onward, you could modify the `moveit_msgs/msg/PlanningOptions.msg` and add a field `reuse_planner_cache` or something like that.

GabrielKoenig, 2019-09-10:
@BryceStevenWilley are there any updates regarding the possibility to maintain planner data?

BryceStevenWilley, 2019-09-11:
@GabrielKoenig I'm gonna echo v4hn here: feel free to file a PR yourself for this! I don't quite have the time to work on this myself, but I'm able to review and help out anyone who wants to try for this.

BryceStevenWilley, 2019-09-13:
SRT and RRF aren't currently available in MoveIt yet (all of MoveIt's sampling-based algorithms are implemented in a separate library, OMPL, which lists it's available planners [here](https://ompl.kavrakilab.org/core/planners.html)), but I think you could find a planner that provides similar functionality to those two (maybe [SPARS](https://ompl.kavrakilab.org/core/classompl_1_1geometric_1_1SPARS.html#gSPARS)?) And I think @v4hn's suggestion of adding a field to `PlanningOptions.msg` is the better bet, otherwise I think you're likely to clear the planner data too often due to noise. You might get familiar with [this class](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L551) as for how to implement it.

simonschmeisser, 2020-03-21:
I think this can be closed @mamoll ?

v4hn, 2020-03-21:
Resolved in https://github.com/ros-planning/moveit/pull/1799

I would still like an official external mechanism to clear the current cache (if we know our environment changed drastically), but that's probably left as future work.

closed 2020-03-21 by v4hn