moveit/moveit #1034: failing unittest on ROS build server

rhaschke, 2018-08-23:
The unittest `MoveItConfigData.ReadingControllers` fails for some reason on the ROS build server,
while there is no issue on Travis:
http://build.ros.org/job/Mdev__moveit__ubuntu_bionic_amd64/47/testReport/(root)/MoveItConfigData/ReadingControllers/
http://build.ros.org/job/Kdev__moveit__ubuntu_xenial_amd64/lastCompletedBuild/testReport/(root)/MoveItConfigData/ReadingControllers/

@MohmadAyman: Could you have a look, please?

MohmadAyman, 2018-08-23:
I believe that have sth to do with the tests trying to write a `ros_controllers.yaml` to `moveit_resources` which is in `/opt`.  
`Unable to open file for writing /opt/ros/melodic/share/moveit_resources/ros_controller.yaml`

I'm guessing Travis has `moveit_resources` built from source thats why it doesn't fail, or run the tests as a `su`.

I'm not sure what is the difficulty of modifying the `ros build server`, well since we only need a temp path for this file, we can try to come up with a path that would work.

v4hn, 2018-08-23:
> `Unable to open file for writing /opt/ros/melodic/share/moveit_resources/ros_controller.yaml`

Nothing should ever try to write to a package path.
If you need a temporary file, it's usually a good idea to use `mkstemp` et al,
but I did not look at the concrete case

davetcoleman, 2018-08-23:
Indeed, the test should write to a temp location. @MohmadAyman can you propose a PR for this?

However this may not be supported [at all by GTest](https://stackoverflow.com/questions/38398288/generate-temporary-file-folder-c-gtest) in which case, maybe we can output the file to memory then read it again from memory without an actual file hit?

MohmadAyman, 2018-08-24:
[This commit](https://github.com/ros-planning/moveit/pull/1024/commits/b20bc98bebd0da25daaeabfd9106bb13af9fe939) should fix this issue.

rhaschke, 2018-08-26:
> This commit should fix this issue.
@MohmadAyman It would have been better to file a separate PR for this.

MohmadAyman, 2018-09-01:
Open pull request #1042 to solve this issue.

davetcoleman, 2018-09-06:
Closed for now by https://github.com/ros-planning/moveit/pull/1042, reopen if build farm still is failing

closed 2018-09-06 by davetcoleman