moveit/moveit #1035: Can move_group load URDF without restart

itfanr, 2018-08-24:
Hello everyone, 

1、Can move_group generate config without assistant UI. I want to use command line or shell instead.
2、Can move_group load URDF without restart?

Thank you.

rhaschke, 2018-08-24:
> 1. Can move_group generate config without assistant UI. I want to use command line or shell instead.

The setup assistant UI is just a convienence helper to generate all those files of a `moveit_config` package. If you have another method to gather all the required information you could also generate those files using a complex shell script. However, I think this essentially will boil down to filling some very simple templates, because there are lots of parameters that are interactively gathered in the GUI. So, I don't see your goal yet.

> 2. Can move_group load URDF without restart?
No, ROS in general doesn't support reloading of URDF on the fly. Remember, it's not only the move_group node, but many other nodes relying on this robot description as well.

closed 2018-08-24 by rhaschke

itfanr, 2018-08-24:
@rhaschke  Thank you for your answer. 

I benefit a lot but I think generate from shell is needed in production environment .

rhaschke, 2018-08-24:
Probably, I don't get your use case yet. Please can you elaborate more?
Usually the configuration is created once (during setup-time of your production environment) and then never changed.

itfanr, 2018-08-24:
@rhaschke  
Our product is  a universle controller that can apply to many kinds of robot arm.
Therefore, we want users customize their robot. It is obvious that  restarting programme can impact the user’s experience.

k-okada, 2019-02-07:
@rhaschke I also interested in this topic, because when the robot changed the gripper (a.k.a tool changer), we want to update the robot model. 

c.f. https://www.youtube.com/watch?v=y64-D6s2HvA
c.f. https://github.com/jacquelinekay/rep/blob/reload_robot_model/rep-reload-robot-model.rst

rhaschke, 2019-02-07:
@k-okada I know about this need, but I don't think that there is a chance to get this feature in ROS1.
There are too many independent packages relying on the robot_description and all of them need to be updated...

jonbinney, 2019-02-07:
Yeah this sounds hard - examples of other nodes that would need to also be able to reload the urdf:
- robot_state_publisher
- rviz (robot_model plugin, for example)
- gazebo (for cases of simulated robot arms)

As @rhaschke says, it seems unlikely to get this into ros1 :-(