moveit/moveit #1036: Moveit Setup Assistant can't parse XML description files

shreyasgan, 2018-08-24:
### Description

[rospack] Error: no package given
[librospack]: error while executing command
[ INFO] [1535132859.993706205] Running 'rosrun xacro xacro.py --inorder /home/bashir/ws_moveit/src/staubli_experimental/staubli_tx90_support/urdf/tx90.xacro

XML parsing error: not well-formed (invalid token): line 27, column 1
when processing file:

Check that:
- Your XML is well-formed
- You have the xacro xmlns declaration: xmlns:xacro= "http://www.ros.org/wiki/xacro"

### Your environment
* ROS Distro: Indigo
* OS Version: e.g. Ubuntu 14.04
* Binary Build

### Steps to reproduce
clone the staubli_experimental package from github: https://github.com/ros-industrial/staubli_experimental
Then run setup assistant and try to open tx90.xacro

### Expected behaviour
Display a model of the TX90 in the setup assistant

### Actual behaviour
I recieve an error

rhaschke, 2018-08-26:
Works perfectly on my side. Did you checked the xacro command on the command line too?

saurabhbansal90, 2018-08-30:
Please try with upgraded MoveIt!

BryceStevenWilley, 2019-06-20:
Cannot reproduce, closing for now.

closed 2019-06-20 by BryceStevenWilley