moveit/moveit #1052: Compilation of moveit_ros/visualization fails because of boost::mutex

timonegk, 2018-09-06:
### Description

Compilation of moveit_ros/visualization for melodic fails because of `boost::mutex`

### Your environment
* ROS Distro: Melodic
* OS Version: Arch Linux
* Source build
* Branch melodic-devel (c2ab8a8feb6d1fca126348d5deb0bd8db1f87a8c) (but also in releases, e.g. 0.10.1, 0.9.9)

### Steps to reproduce
```bash
git clone https://github.com/ros-planning/moveit.git
cd moveit/moveit_ros/visualization
mkdir build
cd build
cmake .. -DPYTHON_EXECUTABLE=/usr/bin/python3
make
```

### Expected behaviour
The package should compile successfully

### Actual behaviour
The package fails because of `boost::mutex`.

### Backtrace or Console output
```In file included from /tmp/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp:40:
/tmp/moveit/moveit_ros/visualization/rviz_plugin_render_tools/include/moveit/rviz_plugin_render_tools/trajectory_visualization.h:142:10: error: ‘mutex’ in namespace ‘boost’ does not name a type
   boost::mutex update_trajectory_message_;
          ^~~~~
/tmp/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp: In member function ‘virtual void moveit_rviz_plugin::TrajectoryVisualization::update(float, float)’:
/tmp/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp:368:12: error: ‘boost::mutex’ has not been declared
     boost::mutex::scoped_lock lock(update_trajectory_message_);
            ^~~~~
/tmp/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp: In member function ‘void moveit_rviz_plugin::TrajectoryVisualization::incomingDisplayTrajectory(const ConstPtr&)’:
/tmp/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp:491:12: error: ‘boost::mutex’ has not been declared
     boost::mutex::scoped_lock lock(update_trajectory_message_);
            ^~~~~
```

Adding `#include <boost/thread/mutex.hpp>` seams to fix the problem. But since I am not very familiar with cpp, this might not be the best solution.
Maybe this is just a problem with my OS, though. I know that Arch is not officially supported.

rhaschke, 2018-09-07:
You should switch to the melodic-devel branch after cloning!

rhaschke, 2018-09-07:
Probably, your issue is related to your OS, indeed.

> Adding #include <boost/thread/mutex.hpp> to moveit_ros/visualization/rviz_plugin_render_tools/include/moveit/rviz_plugin_render_tools/trajectory_visualization.h seams to fix the problem.

Please file a PR adding this missing include.

timonegk, 2018-09-09:
Closed in 83a8a6f2d67bb46712f13a3266687bb17c9724ce

PR #1055 (2018-09-08): Include boost/thread/mutex.hpp in trajectory_visualization.h
### Description

Closes #1052. See the issue for further information.

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
commit 83a8a6f2 (2018-09-08): Include boost/thread/mutex.hpp in trajectory_visualization.h (#1055)

Fixes #1052
closed 2018-09-09 by timonegk