moveit/moveit #1058: Add goal orientation tolerance around three axis in move_group_interface

wanderhappy, 2018-09-10:
### Description

Add the function of set_goal_orientation_tolerance around three axis(x y z) in move_group_interface. Now it can only set the tolerance as a whole. It is really useful when planning a picking task with suckers. If the gripper or sucker is axial symmetry, no need to set the angle around the axis.
### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Expected behaviour
setGoalOrientationTolerance(absolute_x_axis_tolerance, absolute_y_axis_tolerance, absolute_z_axis_tolerance);

### Actual behaviour
setGoalOrientationTolerance(double tolerance)

v4hn, 2018-09-11:
Please provide a patch.

fatihirim, 2018-11-27:
Any plan to get this feature included in moveit soon?

v4hn, 2018-11-29:
As soon as someone writes a patch for it and files a pull-request.

Please feel invited to do that!

It is rather straightforward to add functions for this in
`moveit_core/kinematic_constraints/src/utils.cpp`
and
`moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp`