moveit/moveit #1062: Duration is out of dual 32-bit range

gstavrinos, 2018-09-12:
Dear MoveIt! developers,

### Description

After updating my Ubuntu packages (and later cloned the latest kinetic-devel branch) I keep getting the error ` 'std::runtime_error'  what():  Duration is out of dual 32-bit range` after almost each successful plan execution. The error is encountered both by sending goals from rviz and code.

### My environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build: Both latest

Thanks for any kind of help,

George

gavanderhoorn, 2018-09-12:
The first thing to check is whether you've upgraded *all* your ROS pkgs. So `sudo apt-get upgrade` should not show any ROS pkgs with outstanding updates.

gstavrinos, 2018-09-12:
Sorry my answer was not detailed enough. Before trying to compile my catkin workspace, I tried upgrading but apt found nothing new.

(0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.)

gavanderhoorn, 2018-09-12:
I'm not claiming it is the reason, but because of all the other reports of ABI breaks it was a good thing to check.

You're reporting this issue with both the current `.deb`s as well as a source build, correct?

gstavrinos, 2018-09-12:
Thanks for your help. You are correct, I have the latest binaries and the latest source. I will keep an eye on ROS answers and on the issues here, for candidate solutions.

Thanks again.

skohlbr, 2018-09-12:
To be more precise: A complete, clean rebuild of the workspace is required, not just compiling again. We also ran into this a few days ago.

gavanderhoorn, 2018-09-12:
@skohlbr wrote:
> To be more precise: A complete, clean rebuild of the workspace is required, not just compiling again. We also ran into this a few days ago

Yes, indeed.

However, as the OP reports the same issue with the latest `.deb`s, it seems unlikely this is the same ABI break as reported on ROS Answers (unless the `.deb`s have not been properly rebuilt).

gstavrinos, 2018-09-12:
> To be more precise: A complete, clean rebuild of the workspace is required, not just compiling again. We also ran into this a few days ago.

Thanks for joining in, on the discussion. I followed your advice and just finished compiling with a clean workspace (removed both `build` and `devel` folders). Unfortunately the problem persists.

gavanderhoorn, 2018-09-20:
I'm still not convinced this is not the ABI incompatibility.

We have a large nr of Q&As on ROS Answers about this and it's almost always resolved by updating all packages and rebuilding the workspace.

A simple `catkin b` doesn't work, you'll have to really clean it out first then rebuild.

gstavrinos, 2018-09-21:
I cleaned all of my catkin workspaces, updated all my packages and had no chance. I tried this routine everyday, before starting to work on my regular stuff, and suddenly yesterday I had no crashes. I am not sure if I was lucky or it was actually fixed, but it seems to be fine for now...

otamachan, 2018-09-26:
Hi, just FYI, I encountered the same issue and I found that ABI breaking change was introduced to `rostime` by this [PR](https://github.com/ros/roscpp_core/pull/82). This was released on [Sep 06th](https://discourse.ros.org/t/new-packages-for-kinetic-2018-09-06/5983).

ManojGuha, 2018-11-15:
I got this issue. Tried many things but nothing worked. Finally uninstalled and reinstalled ROS. Now everything is fine

poripsasw, 2019-01-02:
Hi, I have got the same issue but its not consistent. MoveIt retime_trajectory throws this error from time to time (once in 100-300 runs).I have recently installed ros (on a docker) so all packages should be new. I have installed by ubuntu repositories i.e `apt-get install`.

rahulbhadani, 2019-04-03:
I am also encountering the same issue on ROS Melodic running on Ubuntu 18.04.

rhaschke, 2019-04-04:
@rahulbhadani, please follow the instructions in this thread, i.e. cleaning and rebuilding your workspace from scratch.

closed 2019-04-04 by rhaschke

amjack0, 2020-07-14:
I was using 18.04, in my ros node, ros rate was not defined ! Defining it worked.