moveit/moveit #1063: Get joint values from SRDF

keerthanamanivannan, 2018-09-17:
### Description

Get joint values specified in the `group_state` in the SRDF in moveit_config

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary Build
* Binary release version: 0.9.12-Alpha

I realize that this is not an issue, but more of a question. I posted in answers.ros.org and I haven't gotten any replies, thus I am posting here. 

Say, I have three poses of robot described in the robot SRDF. How would I access these values with a moveit_commander instance?

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

keerthanamanivannan, 2018-09-17:
ok, I see that `get_named_target_values` haven't been exposed in [here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_commander/src/moveit_commander/move_group.py). I guess I can expose it and make a PR or something.

rhaschke, 2018-09-18:
Yes, you are welcome to file a PR.

simonschmeisser, 2019-02-14:
@rhaschke @keerthanamanivannan can be closed right?

rhaschke, 2019-02-14:
Yep. The PR is merged. Thanks for checking.

closed 2019-02-14 by rhaschke