moveit/moveit #1064: IK failed, Manipulation plan failed at stage 'reachable & valid pose filter'

AIWintermuteAI, 2018-09-18:
### Description

I'm using MoveIt to control 4-DOF robotic arm I built myself. The first stage is documented here 
https://steemit.com/robot/@wintermuteai/ros-moveit-robotic-arm
I'm at the second stage now, where I'm trying to use Python API for pick and place pipeline. First the arm goes to the pre-grasp pose using set_pose_target command, which arm can execute 80 percent of the time.
But pick operation always fails with following error:

[ WARN] [1537283456.419090736]: All supplied grasps failed. Retrying last grasp in verbose mode.
[ INFO] [1537283456.419349063]: Re-added last failed plan for pipeline 'pick'. Queue is now of size 1
[ INFO] [1537283456.425039540]: IK failed
[ INFO] [1537283456.432710067]: IK failed
[ INFO] [1537283456.441817085]: IK failed
[ INFO] [1537283456.441934449]: Sampler failed to produce a state
[ INFO] [1537283456.442002976]: Manipulation plan 8 failed at stage 'reachable & valid pose filter' on thread 0

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
*  Binary
* If binary, which release version?

### Steps to reproduce
Here's my urdf file 

```
<robot name="test_robot">
  <link name="base_cylinder">
	<visual>
         <geometry>
          <mesh filename="package://urdf_tutorial/meshes/base_cylinder.stl" scale ="0.001 0.001 0.001"/>
         </geometry>
        </visual>
        <collision>
         <geometry>
          <mesh filename="package://urdf_tutorial/meshes/base_cylinder.stl" scale ="0.001 0.001 0.001"/>
         </geometry>
        </collision>
  </link>

  <link name="platform">
	<visual>
         <geometry>
          <mesh filename="package://urdf_tutorial/meshes/top_plate.stl"/>
         </geometry>
        </visual>
        <collision>
         <geometry>
          <mesh filename="package://urdf_tutorial/meshes/top_plate.stl"/>
         </geometry>
        </collision>
  </link>

  <link name="lower_arm">
	<visual>
         <geometry>
          <mesh filename="package://urdf_tutorial/meshes/lower_arm.stl"/>
         </geometry>
        </visual>
        <collision>
         <geometry>
          <mesh filename="package://urdf_tutorial/meshes/lower_arm.stl"/>
         </geometry>
        </collision>
  </link>

  <link name="upper_arm">
	<visual>
         <geometry>
          <mesh filename="package://urdf_tutorial/meshes/top_arm.stl"/>
         </geometry>
[... 560 lines cut ...]
        self.scene_pub.publish(p)

if __name__ == "__main__":
    MoveItDemo()
```
### Expected behaviour
Robot arm going to pre-grasp pose and then grasping an object

### Actual behaviour
Arm going to pre-grasp pose and then failing to execute all the grasps provided by grasp generator. 
A few guesses:
1)I saw people having issues with less than 6-DOF arms using KDL kinematics solver. I've tried Trac-IK with the same results. I also want to pay attention that my arm can move to different poses using RVIZ interactive markers, but the strange thing is, after joint2 or joint3 moved from the default angle I cannot rotate the robot platform(base_cylinder) with interactive markers anymore without swtiching on the "Allow approximate IK solutions" option.
2)There's no IK solver for gripper which is two revolute type joints. Can that be an issue? 

I would appreciate any help, that could move me in the right direction.

### Backtrace or Console output

[ WARN] [1537283456.419090736]: All supplied grasps failed. Retrying last grasp in verbose mode.
[ INFO] [1537283456.419349063]: Re-added last failed plan for pipeline 'pick'. Queue is now of size 1
[ INFO] [1537283456.425039540]: IK failed
[ INFO] [1537283456.432710067]: IK failed
[ INFO] [1537283456.441817085]: IK failed
[ INFO] [1537283456.441934449]: Sampler failed to produce a state
[ INFO] [1537283456.442002976]: Manipulation plan 8 failed at stage 'reachable & valid pose filter' on thread 0

v4hn, 2018-09-18:
(1) You don't have to go to a pre-grasp pose yourself with the pick-pipeline, it should take care of that

(2) Inverse kinematics for your grasp failed, so the grasp you specify cannot be reached by your gripper - or your IK solver at least thinks so

(3) The Pipeline expects to be able to do Cartesian movements along the retreat and approach directions. This might be a problem with only 4DOF (it depends on your request and the arm kinematics of course)

AIWintermuteAI, 2018-09-19:
1) Okay. I saw it in the tutorial and many people seem to specify the pre-grasp pose, so I assumed it's necessary. If not, even better.
2) Any way to check the location of grasps? Maybe with markers in RVIZ?
3)That might be an issue, because as I mentioned arm kinematics does seem to be a little weird with KDL and TracIK plugins. I guess I'll have to generate IKfast plugin... Any idea which kinematics solver to use with 4 DOF? 

    Transform6D - end effector reaches desired 6D transformation
    Rotation3D - end effector reaches desired 3D rotation
    Translation3D - end effector origin reaches desired 3D translation
    Direction3D - direction on end effector coordinate system reaches desired direction
    Ray4D - ray on end effector coordinate system reaches desired global ray
    Lookat3D - direction on end effector coordinate system points to desired 3D position
    TranslationDirection5D - end effector origin and direction reaches desired 3D translation and direction. Can be thought of as Ray IK where the origin of the ray must coincide.
    TranslationXY2D - end effector origin reaches desired XY translation position, Z is ignored. The coordinate system with relative to the base link.
    TranslationLocalGlobal6D - local point on end effector origin reaches desired 3D global point. Because both local point and global point can be specified, there are 6 values.
    TranslationXAxisAngle4D, TranslationYAxisAngle4D, TranslationZAxisAngle4D - end effector origin reaches desired 3D translation, manipulator direction makes a specific angle with x/y/z-axis (defined in the manipulator base link’s coordinate system)

I've tried Transform6D and it says my arm is < 6 DOF....

v4hn, 2018-09-19:
1. You have to *specify* it implicitly in the Grasp message, but you don't have to move the robot there before grasping.

2. We are currently working on [a replacement for this pipeline](https://github.com/ros-planning/moveit_task_constructor). It's not as straight-forward to use as the pick function yet, but visualizes a lot of useful feedback ootb. There will be a talk about it in two weeks at ROSCon (recording will be made available afterwards).

3. This is not a problem in MoveIt anymore, so I will close this issue. Please post a question on [ros answers](https://answers.ros.org) that also includes a picture/kinematic schema of your robot and hopefully someone can help you there.

closed 2018-09-19 by v4hn

smarthhy, 2019-04-04:
@AIWintermuteAI 
Hi,
Have you solved this problem? Thanks

AIWintermuteAI, 2019-04-04:
I went another way and used Grasp Pose Detection (GPD) package for obtaining the grasp pose from point cloud. I improved IK success rate by adding one more degree of freedom. But it still fails in about 20% of cases, which I solve iterating fast through the grasps and finding the one that works

saitejaallam, 2020-07-09:
@AIWintermuteAI 
Hi,
I am using open manipulator mounted on the turtlebot3 waffle pi to perform the pick and place demo. I came across the same issue as you have faced. Here is my output from the console.

> [ WARN] [1594306947.954874050, 116.642000000]: All supplied grasps failed. Retrying last grasp in verbose mode.
[ INFO] [1594306947.955223575, 116.642000000]: Re-added last failed plan for pipeline 'pick'. Queue is now of size 1
[ INFO] [1594306947.963400818, 116.650000000]: IK failed
[ INFO] [1594306947.972309928, 116.654000000]: IK failed
[ INFO] [1594306947.979752280, 116.662000000]: IK failed
[ INFO] [1594306947.980016702, 116.662000000]: Sampler failed to produce a state
[ INFO] [1594306947.980154410, 116.662000000]: Manipulation plan 0 failed at stage 'reachable & valid pose filter' on thread 0
[ INFO] [1594306947.980542898, 116.662000000]: Pickup planning completed after 0.071445 seconds

I am very eager to know the issue and solve it. Here is my code to run this demo.
`#!/usr/bin/env python

import sys
import rospy
from math import pi,radians, sqrt
from tf import transformations
from tf.transformations import quaternion_from_euler, euler_from_quaternion
from moveit_commander import RobotCommander, MoveGroupCommander
from moveit_commander import PlanningSceneInterface, roscpp_initialize, roscpp_shutdown
from geometry_msgs.msg import PoseStamped, Quaternion, Pose
import actionlib
from math import pi, radians, sqrt
from moveit_msgs.msg import Grasp, GripperTranslation, PlaceLocation, ExecuteTrajectoryGoal, ExecuteTrajectoryAction, ExecuteTrajectoryFeedback
from trajectory_msgs.msg import JointTrajectoryPoint

if __name__=='__main__':

    roscpp_initialize(sys.argv)
    rospy.init_node('demo', anonymous=True)
   
    #Use the planning scene object to add or remove objects
    scene = PlanningSceneInterface()
    robot = RobotCommander()

    arm = MoveGroupCommander("arm")
    gripper = MoveGroupCommander("gripper")
    #eef = arm.get_end_effector_link()
    #print "eef:",eef
    rospy.sleep(2)

    #started the action client to perform the action and wait for sometime to execute the subsequent operation
    #otherwise it throws a warning in the console

    arm_client = actionlib.SimpleActionClient('execute_trajectory',ExecuteTrajectoryAction)
    arm_client.wait_for_server()
    rospy.loginfo('Execute Trajectory server is available for arm')
  
    gripper_client = actionlib.SimpleActionClient('execute_trajectory',ExecuteTrajectoryAction)
    gripper_client.wait_for_server()
    rospy.loginfo('Execute Trajectory server is available for gripper')

    gripper_joint_values = gripper.get_current_joint_values()
    gripper_joint_values[0] = 0.018
    gripper_joint_values[1] = 0.0179
    gripper.set_joint_value_target(gripper_joint_values)
    plan = gripper.plan()
    print("plan executed")
    rospy.sleep(1)
    gripper.go(wait=True)
    rospy.sleep(1)

    # clean the scene
    scene.remove_world_object("table")
    scene.remove_world_object("part")

    arm.set_named_target("home")
[... 125 lines cut ...]
    # append the grasp to the list of grasps
    grasps.append(g)

    rospy.sleep(2)

    #interface = MoveGroupInterface("arm", "base_footprint", "move_group")
    #interface.setPlanningTime(45.0)
    result = False
    n_attempts = 0

    # pick the object
    while result == False:

        result = robot.arm.pick("part", grasps)
        n_attempts +=1
        print "Attempts:", n_attempts
        rospy.sleep(0.2)
    #print("gr:", grasps)
    rospy.loginfo("part found and picking")

    rospy.spin()
    roscpp_shutdown()
   `

Please help me in solving this issue. I am able to get the gripper open position and the arm moving to its home location specified in the code.