moveit/moveit #1065: set_max_{velocity,acceleration}_scaling_factor not working in compute_cartesian_path

pingplug, 2018-09-19:
Why use 1.0 (max velocity) there?
https://github.com/ros-planning/moveit/blob/54d516a4e6fdc5db9dfcba4da9e916d807ea312a/moveit_ros/move_group/src/default_capabilities/cartesian_path_service_capability.cpp#L158

v4hn, 2018-09-19:
> Why use 1.0 (max velocity) there?

Because the GetCartesianPath service interface we use with this capability does not have the fields to request user-defined max velocity/acceleration values.

Feel free to file pull-requests for the melodic-devel branches (we are still willing to change message formats there) for moveit_msgs and moveit_ros/planning_interface/move_group_interface (and possibly the python wrappers if they have to be adjusted separately).

There is nothing wrong with adding it.
I still favor a rewrite of the CartesianPath functionality to support it as a generic MoveIt planner (which automatically provides these fields).
But this is a big project and afaik nobody attempted it yet.

Last remark: when people talk about Cartesian trajectories they usually also want to have constant eef velocities.
This is not addressed here at all at the moment, although it is straight-forward to rescale the trajectory w.r.t. the moved distance of the end effector between waypoints.

felixvd, 2019-11-17:
I'd like to suggest this as a WMD issue. This sort of parametric planning is simple, cheap and useful.

This issue could be A) either wrapping a Cartesian planner as a generic MoveIt planner, or B) just exposing the velocity parameters in the current one. 

This is connected to #1556 , which is about expressing the scaling factor as cartesian speed.

v4hn, 2019-11-18:
I do not think we need your proposed solution A, as the [`pilz_industrial_motion`](https://github.com/PilzDE/pilz_industrial_motion) plugin provides similar support already.

B) is clearly a WMD issue and consists of adding scaling parameters to the `GetCartesianPath` service definition and using them to compute the velocity profile in the service.

gautz, 2021-06-23:
Could this be closed in favor of #2674 ?

rhaschke, 2021-06-24:
I added a corresponding link, which will automatically close this issue as soon as #2674 is merged.