moveit/moveit #1067: get_planning_scene with 0-component should return complete scene

v4hn, 2018-09-23:
Currently, the `/get_planning_scene` service returns an empty scene message when the request field has `components` set to 0. I believe it would be more useful if the service would return the *full* planning scene in this case.

It is cumbersome and error-prone to specify each and every bit (i.e. bitmask 1023) explicitly for each full call.

It should be validated throughout MoveIt whether we call the service like this in some context (although it does not make much sense...), because the ROS overhead would increase tremendously.

gavanderhoorn, 2018-09-24:
> It is cumbersome and error-prone to specify each and every bit (i.e. bitmask 1023) explicitly for each full call.

`(~0) & 0x3FF`?

j-petit, 2019-03-31:
I am taking this up.

rhaschke, 2019-05-14:
Addressed in #1424

closed 2019-05-14 by rhaschke