moveit/moveit #1068: Remote center of motion using moveit

saqrhmad7, 2018-09-24:
### Description

I am using 7 DOF Franka robot to perform remote center of motion (RCM) for minimally invasive surgery research.
Actually, I don't know much ways to perform RCM, so I took the easiest way by using a sphere formula to find the coordinates of movement.
I computed the points and use an inverse kinematics solver (TRAC-IK).
The problem is the way the robot moves, not a straight Ecludian way nor a spherical way, it moves in a step way (ex movement on x-axis then on y-axis)
I am thinking of interpolating a lot of points between the current position and the goal, but is there an easier way to control the RCM using moveit in a spherical movement.

Best Regards

### MY environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Robot: Franka Emika 7-DOF

nambakkam, 2020-04-21:
Hi saqrhmad, can you give me an idea how did you implement the RCM?

nihalar, 2020-05-16:
> Hi saqrhmad, can you give me an idea how did you implement the RCM?

Hi, I'm facing the same issue. You have any ideas?

felixvd, 2020-05-17:
1. You can reduce the step size on your interpolation to track the circular trajectory more closely.

2. You can use the CIRC motion defined in the `pilz_industrial_planner` package which is slated to be merged into moveit soon.

3. If the remote center is always the same (e.g. a tool center point), you can add a link to your URDF, set the `end_effector` link to it, and then plan a rotation around it. Also see [here](https://answers.ros.org/question/350967/how-do-i-relocate-the-interactive-marker-to-the-tip-of-the-needle/) on how to set the link and define it as an end effector.