moveit/moveit #1069: select planning group in planning tab

v4hn, 2018-09-24:
For a few years now, it frustrates me that the MotionPlanning display property is the only way to change the active planning group.
The user has to expand the properties of the display, expand the properties of "Planning Request" only to change the planning group property.

With multi-arm setups or setups that model the gripper as a planning group, this is very annoying.

I propose to add a drop-down menu to the "Query" box in the "Planning" tab of the panel right above "Select Start State:".
This should always have the value of the mentioned property, exposing it for easy access.

This issue requires previous insight into RViz programming to finish it in one day.

felixvd, 2018-10-11:
I would even go as far as putting radio buttons underneath the "Plan&Execute" button instead of a dropdown menu. It could be a dropdown when 5+ groups exist, but I believe this is the exception. Radio buttons would save hundreds of clicks.

v4hn, 2018-10-24:
The panel should remain compact and @davetcoleman spent some time in the past making it smaller.
So I'm not a big fan of radio buttons; it might make sense for up to 4 groups.

But then, I work with a robot with 12 planning groups, and thus apparently I'm the exception. :)
I believe your current robot setup should have at least 6 planning groups for your three arms+grippers,
possibly more to use multiple arms at once...

PR #1198 (2018-11-27): select planning group in planning tab and simplify selecting goal and start states
Fixes #1069.

The request consists of multiple commits.

The first one fixes rviz crashing when the planning group is changed while executing a trajectory.
The problem was the destruction of a MoveGroupInterface that also holds the SimpleActionClient used to execute the trajectory. The proposed solution here creates a local shared pointer of the object for execution to ensure that the object is not destroyed asynchronously. 

The other two commits change the gui in the planning tab.
This is the old state of the  [... cut]
closed 2018-11-27 by rhaschke