moveit/moveit #1074: Move It! polar robot configuration not moving second joint

grafoteka, 2018-10-01:
### Description

Hi all! I'm fighting to create a polar robot configuration with Move It. 

The polar robot has 3 joints: rotational, rotational and prismatic.

[image]

I have model it in a xacro file and can move it with problems in Rviz with fake controllers.

[Rviz simulation](https://youtu.be/aMUPI4bxpEo)

Here is the xacro file of the robot (be care to uncomment the lines for the joint with the world) [link](https://github.com/grafoteka/Fotokite_arm/blob/master/arm_description/urdf/RRP.xacro)

And for launch it, this is the command:

    roslaunch arm_description rviz.launch

Then I follow the MoveIt! tutorial for a custom robot ([link](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html)) and generate all the files correctly, but as you can see in the next video, it only moves the first joint (rotational) and the third joint (prismatic), but not the second one (rotational). [Video](https://youtu.be/QGH57xaJD80)

And for launch it, this is the command:

    roslaunch rrp_moveit demo.launch

I would apreciate some help in trying to fix this problem. 

Kinds regards, 
Jorge

EDIT: This is the [workspace](https://github.com/grafoteka/Fotokite_arm) for all the project.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04.1
* Binary built. Installed on September 28, 2018. Don't know how to find the version.

### Steps to reproduce
$roslaunch rrp_moveit demo.launch

### Expected behaviour
The robot should be able to move the 3 joints (rotational, rotational, prismatic).
[RViz with fake controllers - video](https://youtu.be/aMUPI4bxpEo)

### Actual behaviour
It only moves the first (rotational) and the third (primatic) joint.
[Problem with MoveIt! - video](https://youtu.be/QGH57xaJD80)

### Backtrace or Console output

[Console output](https://gist.github.com/grafoteka/82fa5af1465baaa850581edbb793c840)

gavanderhoorn, 2018-10-02:
Cross-post of [Move It! polar robot configuration not moving second joint](https://answers.ros.org/question/304617) on ROS Answers.

grafoteka, 2018-10-02:
> Cross-post of [Move It! polar robot configuration not moving second joint](https://answers.ros.org/question/304617) on ROS Answers.

Hi @gavanderhoorn , I post in both sites because I don't know which should be the correct place. If you think that is better, I can remove one.

grafoteka, 2018-10-02:
I have continue trying to move the robot, and what I have discover is that it can move to any place with the random option:
[image]
But still can't do it with the sphere
[Link to Youtube](https://youtu.be/474T07SVSWc)

mikeferguson, 2018-10-02:
So I think the difference here is that interactive markers (the "sphere") depends on IK, whereas the "random valid" does not. Since you only have 3 DOF, the IK is probably failing since it can't match the orientation of the marker.

grafoteka, 2018-10-02:
Hi @mikeferguson. I have just discover that with the answer ros #95626.

The solution is to add in kinematics.yaml the next line:

 `position_only_ik: True`

So, this is solved!

closed 2018-10-02 by grafoteka