moveit/moveit #1075: Cannot connect to warehouse database

moucrob, 2018-10-03:
### Description

https://answers.ros.org/question/304528/moveit-benchmarking-cannot-connect-to-warehouse-database/

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Binary build desktop full

### Steps to reproduce

`roslaunch panda_moveit_config demo.launch db:=true` and click 'connect' in the motion planning display, if not provided add the motion planning display by clicking on 'add'

### Expected behaviour
Want to get my scene queries tab not empty

### Backtrace or Console output
`
ros.warehouse_ros_mongo: Unable to connect to the database at '127.0.0.1:33829'. If you just created the database, it could take a while for initial setup.
`

VitaminLL, 2020-07-15:
https://answers.ros.org/question/304528/moveit-benchmarking-cannot-connect-to-warehouse-database/

rhaschke, 2020-07-20:
Following the instructions [here](https://github.com/ros-planning/warehouse_ros_mongo#building-from-source) to build the mongo C++ client lib and using the `melodic-devel` branch, I was able to connect to the database without problems also in ROS Kinetic / Ubuntu Xenial.

closed 2020-07-20 by rhaschke

felixvd, 2021-03-29:
More recent instructions: https://github.com/ros-planning/moveit/issues/123