moveit/moveit #1082: PlanningScene::getOctomapMsg/processOctomapMsg do not lock Octomap

simonschmeisser, 2018-10-12:
### Description

We had some crashes here when we called PlanningScene::getOctomapMsg while in another thread the Octomap of another PlanningScene was modified. Turns out there only actually is one Octomap shared between all PlanningScenes that you clone. This is fine in our use case.

However getOctomapMsg (and processOctomapMsg) should then lock the shared Octomap.

https://github.com/ros-planning/moveit/blob/98743e92fcaa8c4a6f2c61f0baedac85614f5fdb/moveit_core/planning_scene/src/planning_scene.cpp#L889

We could have a `dynamic_cast` here to `occupancy_map_monitor::OccMapTree` which has the relevant locks

### Your environment
* ROS Distro: Kinetic and current

wongkaiweng, 2018-10-25:
I am starting to work on this as a part of WMD.

wongkaiweng, 2018-10-25:
@simonschmeisser  I tried your suggestion and here is what happened: 
*  To use `occupancy_map_monitor` in `planning_scene.app`, first I included `moveit_occupancy_map_monitor` in [`CMakeList.txt`](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_core/planning_scene/CMakeLists.txt). I did `catkin_make` and I got a cyclic dependencies error. 
```
"moveit_planning_scene" of type SHARED_LIBRARY
    depends on "moveit_occupancy_map_monitor" (weak)
    depends on "moveit_constraint_samplers" (weak)
    depends on "moveit_planning_request_adapter" (weak)
  "moveit_occupancy_map_monitor" of type SHARED_LIBRARY
    depends on "moveit_planning_scene" (weak)
    depends on "moveit_constraint_samplers" (weak)
    depends on "moveit_planning_request_adapter" (weak)
At least one of these targets is not a STATIC_LIBRARY.  Cyclic dependencies are allowed only among static libraries. 
```
It turns out `moveit_occupancy_map_monitor` depends on `moveit_planning_scene` currently. 
In general, `occupancy_map_monitor` is higher-level than `moveit_core` (where `planning_scene.cpp` sits) so `moveit_core` should not depend on `moveit_occupancy_map_monitor`.

A workaround would be to use `planning_scene_monitor::LockedPlanningSceneRO` or `planning_scene_monitor::LockedPlanningSceneRW` to get your planning scene and octomap at the end. Either of them locks your planning scene so that you have a planning scene with read-only access or read/write access:

https://github.com/ros-planning/moveit/blob/aac8c0de00d5f01c2c3e908b8f4028c84756a920/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h#L599

https://github.com/ros-planning/moveit/blob/aac8c0de00d5f01c2c3e908b8f4028c84756a920/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h#L691

simonschmeisser, 2018-10-26:
Thanks for looking at this @wongkaiweng and I'm sorry to hear that idea did not work out as proposed.

We use your suggestion in our code already but this does not stop the next user from introducing the same bug in her application. We should at least add a warning/hint in the doc string of that function. Are you interested in doing that?

When we discussed this internally some time ago, @svenkrause suggested to actually move these functions to PlanningSceneMonitor as only there the locks are easily available. I dismissed that because in the original design concept octomaps should be local to the PlanningScene just like RobotStates, CollisionObjects etc. Since this is not in fact true for octomap anyway we could indeed deprecate that function in PlanningScene and add it PlanningSceneMonitor. Any opinions?