moveit/moveit #1083: Release Kinetic and Meldoic

davetcoleman, 2018-10-13:
Our last Kinetic release was May 29th nearly 5 months ago. The changes from the GSoC efforts are in and we have World MoveIt! Day fast approaching. I think we should release all three branches @130s 

Indigo-specific discussion: https://github.com/ros-planning/moveit/issues/1000

130s, 2018-10-20:
Sure, I'm planning to make releases to K and M this weekend, unless there's objection (if there is, then please specify ticket numbers that need to be resolved).

rhaschke, 2018-10-21:
@130s Maybe, you should start with Indigo first. For K and M, there a plenty of open pull requests and regressions open:
**Kinetic + Melodic**:
- [ ] #1103 

**Melodic**:
- [ ] #1098 (moveit tutorials + panda release)
- [ ] #1096
- [ ] #1090

**Kinetic**:
- [ ] #1091
- [ ] #1056 ?

Probably, there is more.

130s, 2018-10-21:
Thanks @rhaschke for the list.

Just FYI, I don't mind making multiple releases in the short course of period. So make a release now, then keep making patch releases as soon as a patch is accepted (making a release for many changes is the most time-taking part. Release with smaller changes is less costly IMO). Just saying.

rhaschke, 2018-10-23:
> #1098 is irrelevant to a moveit release in my opinion, its just helping prevent future breakages

Agreed.

> #1096 is not a bug, its just a performance improvement

I think we should get this merged in asap, as this is a huge incompatible API change!
It's not easily possible to pass-in Affine3d types to Isometry3d arguments.

>  Can I go ahead and make a release very soon? Please let us know.

Personally, I don't mind. I'm working on the source tree anyways. Maybe, you should ask people who actually care about API stability, like @v4hn or ROS-industrial folks. But probably they don't want to use the fresh Melodic release yet in any case.

130s, 2018-10-23:
>>    #1096 is not a bug, its just a performance improvement
>
> I think we should get this merged in asap, as this is a huge incompatible API change!

API change should be discussed carefully IMO. Right now without #1096 merged in yet, API is kept, meaning #1096 shouldn't block the release.

To me all is cleared. I plan to make a release today.

130s, 2018-10-23:
(@ros-planning/moveit-maintainers Please hold merge. I've already been running in the release pipeline locally.)

130s, 2018-10-24:
- [x] Update changelog (populate new sections by `catkin_generate_changelog` then manually modify if necessary)
- [x] Tag a new version (utilize `catkin_prepare_release`)
- [ ] Run `bloom`.

Running `bloom` is blocked by https://github.com/ros-planning/moveit/issues/1125 as following.

```
$ rosdep update
$ bloom-release --rosdistro kinetic --track kinetic moveit
:
Creating tag: 'upstream/0.9.13'
:
==> git-bloom-generate -y rosdebian --prefix release/kinetic kinetic -i 0 --os-name ubuntu
Explicitly ignoring package 'moveit_experimental' because it is in the `kinetic.ignored` file.
Explicitly ignoring package 'chomp_motion_planner' because it is in the `kinetic.ignored` file.
Explicitly ignoring package 'moveit_planners_chomp' because it is in the `kinetic.ignored` file.
Generating source debs for the packages: ['moveit_ros_move_group', 'moveit_core', 'moveit_planners', 'moveit_kinematics', 'moveit_simple_controller_manager', 'moveit_controller_manager_example', 'moveit_planners_ompl', 'moveit_ros_planning', 'moveit_ros_manipulation', 'moveit_ros_control_interface', 'moveit_ros_warehouse', 'moveit', 'moveit_fake_controller_manager', 'moveit_plugins', 'moveit_ros_robot_interaction', 'moveit_ros_benchmarks', 'moveit_ros_visualization', 'moveit_ros_planning_interface', 'moveit_commander', 'moveit_ros_perception', 'moveit_setup_assistant', 'moveit_runtime', 'moveit_ros']
Debian Incremental Version: 0
Debian Distributions: ['xenial']
Releasing for rosdistro: kinetic

Pre-verifying Debian dependency keys...
Running 'rosdep update'...
Could not resolve rosdep key 'chomp_motion_planner'
Failed to resolve chomp_motion_planner on ubuntu:xenial with: Error running generator: Failed to resolve rosdep key 'chomp_motion_planner', aborting.
chomp_motion_planner is depended on by these packages: ['moveit_ros_planning']
<== Failed
Some of the dependencies for packages in this repository could not be resolved by rosdep.
You can try to address the issues which appear above and try again if you wish.
```

130s, 2018-10-24:
Once https://github.com/ros-gbp/moveit-release/pull/5 is cleared/merged, I can resume releasing @davetcoleman @v4hn @rhaschke

130s, 2018-10-24:
TLDR; I'll include `moveit_planners_chomp` to be released out for the first time (in addition to `chomp_motion_planner`).

At https://github.com/ros-gbp/moveit-release/pull/5#discussion_r227781794, Robert suggested to do so. Although I haven't found anywhere else so far where its release is discussed, AFAICT no other package depends on it so less harm even if there's an issue with the pkg.

```
$ cd moveit && git log -1
commit b6beb0027c733ad007b2326b15b143e46063903a
Author: David Watkins <DavidWatkins@users.noreply.github.com>
Date:   Tue Oct 23 05:59:13 2018 -0400

    Added plan_only flags to pick and place (#862)
    
    Added optional plan_only flags to pick and place
    
    Added plan_only flag to planGraspsAndPick

$ ack-grep -i moveit_planners_chomp .
moveit_planners/chomp/chomp_interface/test/chomp_moveit.test
3:  <include file="$(find moveit_planners_chomp)/test/rrbot_move_group.launch"/>
5:  <test test-name="chomp_test" pkg="moveit_planners_chomp" type="chomp_moveit_test" time-limit="80.0"/>

moveit_planners/chomp/chomp_interface/test/rrbot_chomp_planning_pipeline.launch.xml
12:     <rosparam command="load" file="$(find moveit_planners_chomp)/test/chomp_planning.yaml"/>

moveit_planners/chomp/chomp_interface/test/rrbot_move_group.launch
9:  <param name="$(arg robot_description)" command="$(find xacro)/xacro --inorder '$(find moveit_planners_chomp)/test/rrbot.xacro'"/>
12:  <param name="$(arg robot_description)_semantic" textfile="$(find moveit_planners_chomp)/test/rrbot.srdf" />
16:    <rosparam command="load" file="$(find moveit_planners_chomp)/test/joint_limits.yaml"/>
35:  <include file="$(find moveit_planners_chomp)/test/rrbot_chomp_planning_pipeline.launch.xml" />
46:  <!--include file="$(find moveit_planners_chomp)/test/moveit_rviz.launch">

moveit_planners/chomp/chomp_interface/CMakeLists.txt
2:project(moveit_planners_chomp)

moveit_planners/chomp/chomp_interface/package.xml
2:  <name>moveit_planners_chomp</name>
```

130s, 2018-10-24:
Now release is blocked by #1127

130s, 2018-10-24:
Kinetic done https://github.com/ros/rosdistro/pull/19344

130s, 2018-10-24:
Melodic done https://github.com/ros/rosdistro/pull/19345

rhaschke, 2018-10-25:
Reopening as the [Kinetic release build failed](http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__moveit_ros_planning__ubuntu_xenial_amd64__binary/86/consoleFull#console-section-15), missing `cmake_modules`. This dependency shouldn't be there anymore. Will fix this.

rhaschke, 2018-10-25:
> Will fix this.

#1134 is merged. pre-release test succeeded on my side.

130s, 2018-10-26:
Thanks @rhaschke for the quick fix. I can run another release session this weekend if it's desirable.

rhaschke, 2018-10-27:
@130s I have done (successful) prerelease tests for Kinetic (702c46b4cd2eb5cd13c51bbca0e8690240737759) and Melodic (ab1273ff886e7a03e3ae72df7ea6b4f7cfee320b). Do you have time to finish the releases?
If not, I could jump in tomorrow.

130s, 2018-10-28:
@rhaschke I'm planning to make releases by Monday morning Pacific. In case it doesn't happen by then please take this over.

130s, 2018-10-29:
Melodic https://github.com/ros/rosdistro/pull/19371

130s, 2018-10-29:
Kinetico https://github.com/ros/rosdistro/pull/19372

130s, 2018-10-29:
@rhaschke or @ros-planning/moveit-maintainers I'd appreciate if you could keep the status posted for new release need.

rhaschke, 2018-10-29:
Sorry, I mentioned you in #1170 only.

130s, 2018-10-29:
@tfoote I want to make one more release to fix [the persistent regression](https://discourse.ros.org/t/preparing-for-kinetic-sync-2018-10-26/6611). It'd be great if you could hold the sync for a few more hours...

tfoote, 2018-10-29:
ok, Kinetic is still rebuilding the last release(moveit and downstream takes several hours to rebuild) so we can get this in if you can get it released before the end of today. The sync will have to go out tomorrow morning.

IanTheEngineer, 2018-10-29:
@130s I merged #1170 after it built. We should be ready to release

tfoote, 2018-10-30:
Ok thanks @130s I'll sync when this rebuild finishes. There's going to be a long tail due to the many layers of moveit downstream dependencies needing to rebuild.

130s, 2018-10-30:
Melodic https://github.com/ros/rosdistro/pull/19381

rhaschke, 2018-10-30:
Reopening, due to: https://github.com/ros-planning/moveit/pull/1160#pullrequestreview-169795669

rhaschke, 2018-10-30:
#1174 fixed this and is merged. @130s You could retrigger a release (once again - sorry).

rhaschke, 2018-10-30:
> Meanwhile if someone can open a ticket to look into why any of our infrastructure doesn't pick up those

See here: https://github.com/ros-planning/moveit/pull/1174#issuecomment-434326955

130s, 2018-11-01:
Melodic https://github.com/ros/rosdistro/pull/19395

rhaschke, 2018-11-01:
@clalancette Could you have a look at http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__moveit_ros_perception__ubuntu_bionic_amd64__binary/23/console? I think this time, the build farm failed for a reason unrelated to MoveIt.

clalancette, 2018-11-02:
Yeah, there's been a bunch of timeouts on the buildfarm recently; I re-kicked the job, looks like it passed now.

rhaschke, 2018-11-02:
Thanks a lot. Looks like the Melodic release finally made it as well.

closed 2018-11-02 by rhaschke