moveit/moveit #1087: Multiple instances of OMPLPlannerManager provoke segfault in OMPL logger

stwirth, 2018-10-15:
### Description

In the constructor of OMPLPlannerManager, an output handler for OMPL's logger is created and assigned to the OMPL library:

https://github.com/ros-planning/moveit/blob/54d516a4e6fdc5db9dfcba4da9e916d807ea312a/moveit_planners/ompl/ompl_interface/src/ompl_planner_manager.cpp#L100-L101

When a second instance of OMPLPlannerManager is created, a second output handler is created and OMPL gets that one for its output. When that second instance goes out of scope, OMPL is left with an invalid pointer for its output handler. Subsequent calls to OMPL by the first instance of OMPLPlannerManager then result in segfaults.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which git commit or tag? b84b1bf9f6e7454e8c0351bc241420a8df89317c

### Steps to reproduce

### Expected behaviour
No segfault.

### Actual behaviour
Segfault.

### Backtrace or Console output

```
Thread 576 "tasks_tests" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffad2ff700 (LWP 8222)]
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007fffaf436f0c in ompl::msg::log (file=0x7fffafa27a50 "/home/stwirth/ompl/src/ompl-release/src/ompl/base/goals/src/GoalLazySamples.cpp", line=87, level=ompl::msg::LOG_DEBUG, 
    m=0x7fffafa27aa0 "Waiting for space information to be set up before the sampling thread can begin computation...") at /home/stwirth/ompl/src/ompl-release/src/ompl/util/src/Console.cpp:132
#2  0x00007fffaf547231 in ompl::base::GoalLazySamples::goalSamplingThread (this=0xf31890) at /home/stwirth/ompl/src/ompl-release/src/ompl/base/goals/src/GoalLazySamples.cpp:87
#3  0x00007fffaf549475 in std::_Mem_fn_base<void (ompl::base::GoalLazySamples::*)(), true>::operator()<, void>(ompl::base::GoalLazySamples*) const (this=0xf4b610, __object=0xf31890) at /usr/include/c++/5/functional:600
#4  0x00007fffaf549409 in std::_Bind_simple<std::_Mem_fn<void (ompl::base::GoalLazySamples::*)()> (ompl::base::GoalLazySamples*)>::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=0xf4b608) at /usr/include/c++/5/functional:1531
#5  0x00007fffaf549310 in std::_Bind_simple<std::_Mem_fn<void (ompl::base::GoalLazySamples::*)()> (ompl::base::GoalLazySamples*)>::operator()() (this=0xf4b608) at /usr/include/c++/5/functional:1520
```

jonbinney, 2018-10-17:
@stwirth that wouldn't work with multiple threads, right? It assumes that while an `OMPLPlannerManager` is in context, that is in the only thread using OMPL? Consider the following case:

OMPLPlannerManager A created (stores old output handler and sets new one)
OMPLPlannerManager B created(stores output handler that was set by A and sets new one)
OMPLPlannerManager A destroyed (restores handle that existed at the start; could be the `ompl::msg::OutputHandlerSTD` if nothing else had been set before)
OMPLPlannerManager B destroyed (restores the one that A had set)

This seems a bit messy - another approach would be to assume that within a moveit compiled executable we always want to use the ROS-specific OutputHandler defined here: https://github.com/ros-planning/moveit/blob/54d516a4e6fdc5db9dfcba4da9e916d807ea312a/moveit_planners/ompl/ompl_interface/src/ompl_planner_manager.cpp#L71

Could we have a globally defined singleton OutputHandler that gets created when the process starts, and never gets changed? That would avoid the problems of swapping them out and remembering what was there previously.

stwirth, 2018-10-18:
@jonbinney you are right, my proposal only works if destruction occurs in reverse order of construction. A singleton output handler sounds like it could work. Another solution would be to change the OMPL library to allow a set of output handlers and register and de-register on construction and destruction, respectively. Something like
```C++
ompl::msg::addOutputHandler(ompl::msg::OutputHandler *oh);
ompl::msg::removeOutputHandler(const ompl::msg::OutputHandler *oh);
```

jonbinney, 2018-10-25:
I'll take a stab at fixing this during world moveit day today.