moveit/moveit #1088: 14 DOF: Unable to connect to ROS service client with name: /solve_ik

grafoteka, 2018-10-15:
### Description

I have design a 14 DOF robotic arm and follow the setup moveit tutorials. But when I launch the demo.launch I get the next error:

`Unable to connect to ROS service client with name: /solve_ik`

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source

### Steps to reproduce
Here is the [repo](https://github.com/grafoteka/Fotokite_arm) of the project. You anly have two launch the next:

`$ roslaunch fotokite_two_modules_moveit demo.launch `

### Expected behaviour
You should be able to move the robot to anywhere with RViz and sphere in the EEF.

### Actual behaviour
The IK cannot be solved. I think that the problem is that the IK solvers in moveit only can compute for a maximum of 7DOF.

Maybe, one possible solution could be to create 1 arm of 7-DOF and a second one, attached to the first one with also 7-DOF?

But, how I connect them? I cannot use two virtual joints.

### Backtrace or Console output

[terminal output](https://gist.github.com/grafoteka/110379e5f7e754c993416e174bd5df35)

gavanderhoorn, 2018-10-16:
> The IK cannot be solved. I think that the problem is that the IK solvers in moveit only can compute for a maximum of 7DOF.

A bit pedantic, but all solvers "in MoveIt" are plugins, so technically they are not part of it.

We provide a few default ones (and integrate with a few) but that is something else.

> Maybe, one possible solution could be to create 1 arm of 7-DOF and a second one, attached to the first one with also 7-DOF?

That should not be necessary.

And I actually believe your problem is caused by something else:

> ```
> [ INFO] [1539641786.002969630]: SrvKinematicsPlugin initializing
> [ INFO] [1539641786.005485426]: Loading robot model 'arm'...
> [ INFO] [1539641786.005707248]: Dimension planning group 'arm': 14. Active Joints Models: 14. Mimic Joint Models: 0
> [ INFO] [1539641786.007313907]: waitForService: Service [/solve_ik] has not been advertised, waiting...
> ```

You configured the `arm` group to use the `cached_ik_kinematics_plugin/CachedSrvKinematicsPlugin` ([here](https://github.com/grafoteka/Fotokite_arm/blob/26d93fe28e8e7bced45acec7b977338a36a3be16/fotokite_two_modules_moveit/config/kinematics.yaml#L2)). That plugin requires an *external service server* that is capable of solving IK requests for that group (see [here](https://github.com/ros-planning/moveit/blob/b84b1bf9f6e7454e8c0351bc241420a8df89317c/moveit_kinematics/srv_kinematics_plugin/src/srv_kinematics_plugin.cpp#L125-L143) for the initialisation code). It does not solve any queries itself.

Apparently you don't have such a service server running, so the plugin just reports that it can't find the required service and then tells you that it cannot do any work for you.

If you don't want to use an external IK solver over services, then please configure a different plugin.

The fact that you have a 14 dof setup does not make any difference here.

grafoteka, 2018-10-16:
Hi @gavanderhoorn, 
you were right! Now, I have change to KDL solver, and I can moveit through  a node, but not manually in RViz.

closed 2018-10-16 by grafoteka