moveit/moveit #1105: moveit gives many warnings and failed while executing in rviz

alxer, 2018-10-20:
### Description
Hi~
I try to use Fanuc industry robot to output a simple trajectory through moveit, the terminal gives me following warnings:
[ WARN] [1540042673.043170336, 8.779000000]: Dropping first 1 trajectory point(s) out of 14, as they occur before the current time.
First valid point will be reached in 0.719s.
[ WARN] [1540042676.810792933, 12.531000000]: Controller fanuc_m10ia/fanuc_m10ia_controller failed with error code GOAL_TOLERANCE_VIOLATED
[ WARN] [1540042676.810878284, 12.531000000]: Controller handle fanuc_m10ia/fanuc_m10ia_controller reports status ABORTED
[ INFO] [1540042676.862531475, 12.582000000]: ABORTED: Solution found but controller failed during execution
It is strange that gazebo executes successfully, but the rviz always failed whatever the goal state I set.
### My environment
* ROS Indigo
*  Ubuntu 14.04

Here is the goal state I want to set, I plan and execute it, however, as you can see the commands column, the rviz failed
[image]
Now we have a look at gazebo, it executes successfully.
[image]
The terminal shows in the following way
[image]

I am totally confused, how can I solve this? thanks

AnastasiyaRybakova, 2019-05-07:
Hi, dear. I have faced the same failed in Rviz and also there is no changes in Gazebo. Could You tell me how did You fix this problem?
Thank You.

Deastan, 2019-05-28:
Same issue... I am also working on ROS indigo on ubunutu 14.04. For me I don't care to have an issue with RVIZ... I am simulating on Gazebo and I would like to have the right feedback from the trajectory execution to know if the trajectory was done or not.

`[ WARN] [1559049192.955753441, 25.479000000]: Dropping first 1 trajectory point(s) out of 20, as they occur before the current time.
First valid point will be reached in 0.466s.
[ WARN] [1559049193.008082504, 25.513000000]: Controller PositionJointInterface_trajectory_controller failed with error code PATH_TOLERANCE_VIOLATED
[ WARN] [1559049193.008163096, 25.513000000]: Controller handle PositionJointInterface_trajectory_controller reports status ABORTED
[ INFO] [1559049193.255872832, 25.678000000]: ABORTED: Solution found but controller failed during execution
`

Thank you in advance for anyone whi could help us.