moveit/moveit #1106: [Indigo] pre-release test fails "class planning_scene_monitor::PlanningSceneMonitor’ has no member named ‘waitForCurrentRobotState'"

130s, 2018-10-21:
### Description

Ran a pre-release test locally for [Indigo v0.7.14](https://github.com/ros-planning/moveit/releases/tag/0.7.14) then got the following error in test section.

```
$ docker login    (using your account on hub.docker.com)
$ _DIR_PRLTEST=/tmp/prerelease_job_ind-tru; mkdir -p $_DIR_PRLTEST && cd $_DIR_PRLTEST
$ generate_prerelease_script.py   https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml indigo default ubuntu trusty  amd64  moveit   --level 0  --output-dir ./
$ ./prerelease.sh
:
==> Processing catkin package: 'moveit_ros_move_group'
==> Building with env: '/tmp/catkin_workspace/devel_isolated/moveit_planners_ompl/env.sh'
==> cmake /tmp/catkin_workspace/src/moveit/moveit_ros/move_group -DCATKIN_DEVEL_PREFIX=/tmp/catkin_workspace/devel_isolated/moveit_ros_move_group -DCMAKE_INSTALL_PREFIX=/tmp/catkin_workspace/install_isolated -DCATKIN_ENABLE_TESTING=1 -DCATKIN_SKIP_TESTING=0 -DCATKIN_TEST_RESULTS_DIR=/tmp/catkin_workspace/test_results -G Unix Makefiles in '/tmp/catkin_workspace/build_isolated/moveit_ros_move_group'
-- Boost version: 1.54.0
:
Scanning dependencies of target moveit_move_group_default_capabilities
[ 33%] Building CXX object CMakeFiles/moveit_move_group_default_capabilities.dir/src/default_capabilities/move_action_capability.cpp.o
/tmp/catkin_workspace/src/moveit/moveit_ros/move_group/src/default_capabilities/move_action_capability.cpp: In member function ‘void move_group::MoveGroupMoveAction::executeMoveCallback(const MoveGroupGoalConstPtr&)’:
/tmp/catkin_workspace/src/moveit/moveit_ros/move_group/src/default_capabilities/move_action_capability.cpp:63:38: error: ‘class planning_scene_monitor::PlanningSceneMonitor’ has no member named ‘waitForCurrentRobotState’
   context_->planning_scene_monitor_->waitForCurrentRobotState(ros::Time::now());
                                      ^
make[2]: *** [CMakeFiles/moveit_move_group_default_capabilities.dir/src/default_capabilities/move_action_capability.cpp.o] Error 1
make[1]: *** [CMakeFiles/moveit_move_group_default_capabilities.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'moveit_ros_move_group':
  Command '['/tmp/catkin_workspace/devel_isolated/moveit_planners_ompl/env.sh', 'make', '-j1']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /tmp/catkin_workspace/build_isolated/moveit_ros_move_group && /tmp/catkin_workspace/devel_isolated/moveit_planners_ompl/env.sh make -j1

Command failed, exiting.
# END SUBSECTION                          
```

### Your environment
* ROS Distro: Indigo, Ubuntu 14.04
  ```
  $ apt-cache policy python3-ros-buildfarm                                                                                                                        
  python3-ros-buildfarm:                                                                                                                                                                                      
    Installed: 2.0.1-100                                                                                                                                                                                      
    Candidate: 2.0.1-100                      
  ```
* Source build
* If source, which git commit or tag?: `0.7.14` tag

### Steps to reproduce

See the output above.

### Expected behaviour

Reaches the end of the `prerelease.sh` script.

rhaschke, 2018-10-21:
I will have a look into this.

rhaschke, 2018-10-21:
@130s Hm, this is weird. I cannot find the issue.
[Both travis builds](https://travis-ci.org/ros-planning/moveit/builds/444274938) succeeded with compiling and testing this code - the only exception is `gtest-mesh_filter_test.xml`, which is known to be broken and unrelated. Will this failing test cause issues on the build farm too?

To track down this build issue, can you provide verbose build output?
Looks like the testing phase uses different includes than the build phase. 
Maybe a build order issue? Should there be a `test_depends` in `move_group/package.xml`? 
What is the actual build order?
Are the headers installed at all? I cannot find an `install(DIRECTORY include/ DESTINATION include)`
in most of moveit_ros/* folders. But, that's the same for Kinetic and Melodic and probably was like this forever.

@v4hn Could you please have a look as well? You still have an Indigo setup, don't you?

rhaschke, 2018-10-22:
I've got an Indigo docker container running now. First thing, I observed is a circular package dependency: MoveIt defines a meta package `moveit_full_pr2`, which depends on:
`ros-indigo-moveit-pr2` -> `ros-indigo-pr2-moveit-plugins` -> `ros-indigo-moveit-core`
which explains that all MoveIt binary packages are installed in order to build MoveIt...

rhaschke, 2018-10-22:
OK, running the full pre-release test (without skipping _"build and install"_), I can reproduce the error.

rhaschke, 2018-10-22:
I think, the problem is indeed related to the circular dependency induced by `moveit_full_pr2`. Additionally, I believe there is a bug in catkin's [`list_insert_in_workspace_order`](https://github.com/ros/catkin/blob/kinetic-devel/cmake/list_insert_in_workspace_order.cmake) which is used to order `catkin_INCLUDE_DIRS` [here](https://github.com/ros/catkin/blob/kinetic-devel/cmake/catkinConfig.cmake.in#L108L109). I believe, this function should order dirs in reverse workspace order. Currently we get:

- before ordering: `/tmp/catkin_workspace/install_isolated/include;/usr/include;/usr/include/eigen3;/opt/ros/indigo/include`
- after workspace ordering: `/opt/ros/indigo/include;/tmp/catkin_workspace/install_isolated/include;/usr/include;/usr/include/eigen3`
- where we should get:
`/tmp/catkin_workspace/install_isolated/include;/opt/ros/indigo/include;/usr/include;/usr/include/eigen3`

Because `/opt/ros/indigo/include` comes first, the build first finds the old includes...
I'm not sure, how this could work before...

As a quick fix, I suggest to drop the meta-package `moveit_full_pr2`.
For the future, @dirk-thomas should have a look at the potential catkin bug and give his opinion.
Currently I'm running another pre-release test to validate the approach.

dirk-thomas, 2018-10-22:
> For the future, @dirk-thomas should have a look at the potential catkin bug and give his opinion.

The functionality in `catkin` is backed by a Python function in `catkin_pkg` (https://github.com/ros-infrastructure/catkin_pkg/blob/58df8ca998e56661b700c489c6c6b7051f820e5e/src/catkin_pkg/workspaces.py#L71). The unit test implies that the function does what it is asked to do correctly (https://github.com/ros-infrastructure/catkin_pkg/blob/58df8ca998e56661b700c489c6c6b7051f820e5e/test/test_workspaces.py#L38).

If you still suspect a bug in the CMake function in `catkin` then I would suggest to reproduce the problem and debug the involved bits and pieces, e.g. the arguments / environment when [order_paths.py](https://github.com/ros/catkin/blob/e401545801d261aab46bfb4d93f5981b8acd2dc3/cmake/order_paths.py) is invoked.

rhaschke, 2018-10-23:
New pre-release test succeeded.