moveit/moveit #1110: Lightweight Task-Motion-Planning (Correlated/consecutive/dependent motions planning) - New Feature

Danfoa, 2018-10-22:
### Description

Hey, as suggested in the [contibuting tips](http://moveit.ros.org/documentation/contributing/) (and with the upcoming World MoveIt Day), I use this issue to formally ask if the addition of a light-weight task-oriented-motion planning functionality is a "good idea". **What I want to provide is a simple and higher level approach for planning of motions that are correlated/consecutive/dependent.**

So at work, I stumble with the planning of different manipulation tasks where the robot should grab an object and then perform a specific operation (most likely with kinematic constraints), which sometimes are only possible if the initial robot state (joint configuration) with which the robot approaches the object has certain characteristics, i.g. After the robot takes a bottle with the elbow up configuration, It will crash while attempting to open it, because of some sort of physical constraint that was not taken into account in the approach motion.

So I understand as described [here](https://answers.ros.org/question/274209/planning-together-multiple-sequential-motions-in-moveit/) that multiple motion planning can be performed currently by using a `Plan` instance and the start robot state of the `MoveGroupInterface`, but I believe this seems to be a function that should be simpler to use.

I have already coded a similar function and I found my self using it constantly, and I thought somebody might find it also useful.

What I propose is a simple lightweight interface that uses the already planning framework and would not disrupt the API; Combing the planning of task steps with an analysis of the possibly multiple step start/end robot states/configurations. 

What I have though and roughfly implemented is: 
#### What the end user should provide?
1. A task segmented by steps, which are composed by:
    a. A setp name. i.g. `pouring`, `shaking`, `flipping` (So far I believe this will help make the logging process clear, by giving feedback of which step is the "problematic" one)
    b. A set of `Poses` or `RobotStates`, indicating the key points of the task step.
    c. Kinematic constrained if any shall be applied to the task step.
    d. A scene object and a robot frame, if an object is attached during this task.

The idea would be that the user uses the same type of classes and objects it is already using (i.g. Pose, constrains, JointModels ...) but be able to obtained a higher level approach to multple related motions planning.
#### What the feature should do?
There is literature on many way to do whis, but initially the software should use generic IK plugin functions to obtain a set of valid robot states for each task step poses (start and end), discard configuations that would make the following step unfeasible, and connect the steps through a plan that allows all steps completition.
#### What the feature should return? 
An instance of a `Plan` with a `_trajectory` which satisfies and covers all the task steps requirements

gavanderhoorn, 2018-10-23:
Only leaving this here as it seems highly related (not to say anything about whether or not a contribution like you describe would be "ok" on WMD):

[MoveIt! Task Planning](https://vimeo.com/293432325) ([slides](https://roscon.ros.org/2018/presentations/ROSCon2018_MoveitTaskPlanning.pdf)):

> The MoveIt Task Constructor (MTC) is a recent extension of MoveIt to enable manipulation task planning. It provides a system to design, visualize and plan task-level robot trajectories composed of multiple interdependent sub-stages. In contrast to the previous pick and place manipulation module, it supports users in setting up custom task descriptions in few lines and provides debugging and visualization support along the way.

gavanderhoorn, 2018-10-23:
And I'm sure @v4hn and @rhaschke will add something as well.

felixvd, 2018-10-23:
Such as the [link to the repo](https://github.com/ros-planning/moveit_task_constructor/tree/master)

gavanderhoorn, 2018-10-23:
> Such as the [link to the repo](https://github.com/ros-planning/moveit_task_constructor/tree/master)

which is the one this issue was posted to, is it not?

felixvd, 2018-10-23:
Unless my grasp on reality is fading, this issue is on /ros-planning/moveit, and I linked to the /ros-planning/moveit_task_constructor, which the talk that you linked is about.

gavanderhoorn, 2018-10-23:
You're right, I was under the impression that MTC was already merged into the main repository.

Danfoa, 2018-10-23:
Thank you very much, I had not found this repository or info about this project so far, maybe because the presentation was not so long ago.

I see now that what they are doing is a much more complete idea of what I had thought of.

Thank you very much for your answers.
PS: I will try to find a way to contribute to this project if I am allowed, or if the developers want some support I would be honored to help in any way.

closed 2018-10-23 by Danfoa

rhaschke, 2018-10-23:
@Danfoa Hm. Looks like my answer from last night is somehow lost. github had issues yesterday...
Nevertheless, of course you are very welcome to contribute to this project. To start with, I suggest you start with the demos at https://github.com/ubi-agni/mtc_demos, try to adapt them to your use cases and report about usability.
If you still want to contribute on WMD, we are happy to see some tutorials for this new package submitted to https://ros-planning.github.io/moveit_tutorials resp. https://github.com/ros-planning/moveit_tutorials.

rhaschke, 2018-10-24:
@Danfoa One next important step for MTC is to provide a drop-in-replacement for MoveIt's pick-and-place capability (and a corresponding updated of the [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pick_place/pick_place_tutorial.html)). If you are interested in doing this, please inform us.

Danfoa, 2018-10-24:
@rhaschke Yes I could do that, I have been reading the [core package](https://github.com/ros-planning/moveit_task_constructor/tree/master) code to try to understand how you guys structured the programming, I will have another look after reviewing again your ROSCon presentation, but I believe is easy to understand. Do you already know roughly what feature should show and not show the tutorial? 
PS: I want to be on board, but should we keep discussing this on this issue?

v4hn, 2018-10-24:
@Danfoa We are very happy to get support for the MTC project.
To answer your question: no, let's discuss more contributions off the list,
please contact Robert and me via mail if you want to get involved more.

An actionable for tomorrow could be: look at the demos,
also https://github.com/TAMS-Group/mtc_pour and https://github.com/TAMS-Group/tiago_bartender_mtc ,
and try to setup a reasonably simple pipeline for a robot you are familiar with.
During the process, write down problems you encounter and file an issue with the assorted list.
This would help us quite a bit.

rhaschke, 2018-10-25:
@Danfoa I just [improved the Travis tests](https://travis-ci.org/ros-planning/moveit_task_constructor/builds/445944766). Now it should work... But it doesn't. 

MoveIt's latest melodic release is missing a change from Kinetic that we need.
However, this night we just made a new Melodic release, such that this issue should be gone in the next days (when the release is accepted and build by the build farm). Until then, you need to build MoveIt from source as well.