moveit/moveit #1113: Using stomp with ur5 error

WwYyFan, 2018-10-23:
### Description

When I using stomp with ur5 , terminal output error
```
[ERROR] [1540203992.012618576]: Number of parameters 6 differs from what was preallocated
[ERROR] [1540203992.012642735]: Failed to generate noisy parameters at iteration 1
```
my stomp_planning.yaml file
```
stomp/manipulator:
  group_name: manipulator
  optimization:
    num_timesteps: 60
    num_iterations: 40
    num_iterations_after_valid: 0
    num_rollouts: 30
    max_rollouts: 30
    initialization_method: 1 #[1 : LINEAR_INTERPOLATION, 2 : CUBIC_POLYNOMIAL, 3 : MININUM_CONTROL_COST]
    control_cost_weight: 0.0
  task:
    noise_generator:
      - class: stomp_moveit/NormalDistributionSampling
        stddev: [0.05, 0.8, 1.0, 0.8, 0.4, 0.4, 0.4]
    cost_functions:
      - class: stomp_moveit/CollisionCheck
        collision_penalty: 1.0
        cost_weight: 1.0
        kernel_window_percentage: 0.2
        longest_valid_joint_move: 0.05
    noisy_filters:
      - class: stomp_moveit/JointLimits
        lock_start: True
        lock_goal: True
      - class: stomp_moveit/MultiTrajectoryVisualization
        line_width: 0.02
        rgb: [255, 255, 0]
        marker_array_topic: stomp_trajectories
        marker_namespace: noisy
    update_filters:
      - class: stomp_moveit/PolynomialSmoother
        poly_order: 6
      - class: stomp_moveit/TrajectoryVisualization
        line_width: 0.05
        rgb: [0, 191, 255]
        error_rgb: [255, 0, 0]
        publish_intermediate: True
        marker_topic: stomp_trajectory
        marker_namespace: optimized
``` 
Can someone tell me how to slove it ?

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04

simonschmeisser, 2018-10-24:
Can you please provide a bit more information where your version of stomp comes from? I think it has not officially been released as deb package so you probably installed it from source? Which repository did you use? Can you provide the output of `wstool info` as well?

What exactly did you try to do? Pose goal? JointPosition goal?

migueelnery, 2019-05-16:
I'm having this same error when i try to use Stomp with ur5.
Have you figured out how to solve it?

cardboardcode, 2019-09-19:
Hi there. @WwYyFan @simonschmeisser @BryceStevenWilley  @migueelnery 

Facing the same issue as well when using **STOMP** with **UR5**.

The version of stomp I used came from [ros_industrial/stomp_ros](https://github.com/ros-industrial/stomp_ros). It has been built from source. 

Running `wstool info --target-workspace=stomp_ros` outputs saying there is no workspace configuration.

## [**Environment Specifications**]
```
1. ROS Kinetic Kame
2. Ubuntu 16.04 Xenial Xerus
```
The **STOMP** planner has been set-up, following instructions from the [STOMP planner Moveit tutorial](https://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html).

Here's a summary of steps taken in following the instructions:

1. Built **Moveit** from source in my catkin workspace 
[Guide](https://moveit.ros.org/install/source/)
2. Built my own robot moveit config ROS package 
[Guide](https://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html)
3. Built **stomp_ros** from source instead of the recommended **ros-industrial/industrial_moveit package**. 
[**Rationale**]: Integrating STOMP requires an additional package called `stomp_moveit` which houses the STOMP planner. Neither the **Moveit** nor **the recommended industrial_moveit package** has that. Therefore, running roslaunch when bringing up the robot with the planner throws an error in calling the STOMP planner. 
4. Downloaded and placed `stomp_planning_pipeline.launch.xml` into the **moveit_config** package's `launch` folder. Modified accordingly.
5. Downloaded and placed ` stomp_planning.yaml` into the aforementioned package's `config` folder.
6. Modified `move_group.launch` accordingly.

## [**Further Actions Taken To Reproduce Error**]

Bringing up the robot in simulation shows **STOMP** has been initialized properly. I am using RViz GUI to show the simulation.

Below is a gif screenshot of actions taken in the `MotionPlanning` window of RViz.
[image]

It produces the issue mentioned at the start of the thread.
[image]

Here's the textual abstract of the error:

>[ERROR] [1568852198.587017566]: Number of parameters 6 differs from what was preallocated 
>[ERROR] [1568852198.587044303]: Failed to generate noisy parameters at iteration 1
>[ERROR] [1568852198.587057495]: STOMP failed to find a valid solution after 1 iterations

Been googling for the last week regarding this issue. Not sure if I using the wrong keywords but I am unable to find any solution. So far this is the only thread talking about this so I will post it here. 

#### **Any help is appreciated. Let me know if more info is needed.**

simonschmeisser, 2019-09-19:
Hi @cardboardcode 

stomp got moved to its own repository at https://github.com/ros-industrial/stomp_ros and is no longer part of `industrial_moveit`. Please open an issue at `moveit_tutorials` or maybe even a PR?

Your robot is a 6 degree of freedom (dof) robot (ie has 6 joints). The config you copied is for a Franka robot with 7 dof. Please change this line to contain only 6 entries:

https://github.com/ros-planning/panda_moveit_config/blob/549ea9070fdaefd26b70678ee7651a9f8f689718/config/stomp_planning.yaml#L14

This line determines the amount of noise added to each joint position during rollout. Please also open an issue on `moveit_tutorials` to better explain this. If you are into opening issues you can also open one at https://github.com/ros-industrial/stomp_ros asking for more self explaining error messages.

---

Side remark 1: when running `wstool info` in the source directory of your workspace it should give you a list of all code (url, branch etc) in that folder

2: if you upload your setup (the folder created by setup assistant) to a github repository it is easier to follow and look around

cardboardcode, 2019-09-19:
Hi @simonschmeisser 

Thank you for the prompt reply. 

I will modify the config file as suggested and update if it worked here. 

Will open this issue on **moveit tutorials** as well as **stomp_ros** over the weekend. Once solved, will reference those links here so it is easier for people to find.

cardboardcode, 2019-09-20:
Hi all, 

While the bigger problem I am facing is still present, @simonschmeisser's suggestion resolved the following error which is the main concern of this thread.

> [ERROR] [1568852198.587017566]: Number of parameters 6 differs from what was preallocated
> [ERROR] [1568852198.587044303]: Failed to generate noisy parameters at iteration 1
> [ERROR] [1568852198.587057495]: STOMP failed to find a valid solution after 1 iterations

# [**Solution**]
The following is what I changed specifically to resolve this error when following the [Moveit Tutorial on STOMP Planner](https://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/stomp_planner/stomp_planner_tutorial.html).

#### **[Before]**
In **line 14** of  `stomp_planning.yaml` file. [Reference](https://github.com/ros-planning/panda_moveit_config/blob/549ea9070fdaefd26b70678ee7651a9f8f689718/config/stomp_planning.yaml#L14)
>stddev: [0.05, 0.8, 1.0, 0.8, 0.4, 0.4, 0.4]

#### **[After]**
Same line, same file. Change it to:
>stddev: [0.05, 0.8, 1.0, 0.8, 0.4, 0.4]

Hope this helps whoever has this same issue in the future.