moveit/moveit #1114: Create "Related projects" webpage

davetcoleman, 2018-10-23:
Re-post of @v4hn's issue for improved visibility from https://github.com/ros-planning/moveit.ros.org/issues/164:

> It would be nice to have a page collection of related code projects/plugin on the website.
Some random packages that come to my mind off the top of my head:
https://github.com/ros-industrial/industrial_moveit
https://github.com/TAMS-Group/move_group_online_collision_predictor
https://github.com/TAMS-Group/bio_ik
https://bitbucket.org/traclabs/trac_ik
https://github.com/davetcoleman/moveit_grasps
https://github.com/ros-planning/moveit_visual_tools
https://github.com/ros-planning/moveit_task_constructor
https://github.com/ros-industrial-consortium/descartes

npd26, 2018-10-24:
Also for this,
https://github.com/TAMS-Group/moveit_gpd_pick_object
Tutorial for how to integrate GPD with Moveit?

v4hn, 2018-10-24:
> https://github.com/TAMS-Group/moveit_gpd_pick_object
> Tutorial for how to integrate GPD with Moveit?

More of an example implementation than a "tutorial", but yes.

sharronliu, 2018-10-25:
Added "moveit_gpd_pick_object".

bmagyar, 2018-10-25:
Should we add `ros_control` here too?

sharronliu, 2018-10-26:
> ros_control

added, thanks!