moveit/moveit #1116: Document robot padding and add sample config to Setup Assistant

simonschmeisser, 2018-10-24:
MoveIt! has some capabilities for adding padding in between Robot and Scene Objects. Also per Link and for attached objects. There does not seem to be any documentation/tutorial however and it does not seem to be setup/prepared in MoveIt! Setup Assistant

Look here for a related commit:
https://github.com/ros-planning/moveit/commit/c949b69b3e473e754473f93b1a53be00cc6fbfb1

This could be handled at World MoveIt! Day (I'm not claiming it!)

v4hn, 2018-10-24:
This involves a lot of digging through the code first of all, to understand (and document!) *what* is supported *where* at the moment. If I remember correctly this functionality was not implemented on all levels of the framework.

It would be  very valid outcome of this work package if we had some new issue(s) that point to specific code lines and ask for better support/new interfaces in specific places.

ntwong0, 2019-11-21:
RE: documentation, there's some mention on padding in a couple sections of the tutorial - 

[Planning Scene Tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/planning_scene/planning_scene_tutorial.html) mentions that self-collision checks are unpadded ( relatedly, #683 is a feature request for padding ), and