moveit/moveit #1119: Normalize Quaternions

v4hn, 2018-10-24:
The PlanningScene class does not always correctly normalize quaternions in new collision objects.

Here, the unnormalized w-component is normalized correctly:
```
$ rostopic pub -1 /collision_object moveit_msgs/CollisionObject "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: 'world'
id: 'box'a
primitives:
- type: 1
  dimensions: [1.8,1.5,1.0]
primitive_poses:
- position: {x: 0.0, y: 0.0, z: 0.15}
  orientation: {x: 0.0, y: 0.0, z: 0.0, w: 2.0}"

$ rosservice call /get_planning_scene "components:
  components: 16" | grep box -A20
        id: "box"
        type:
          key: ''
          db: ''
        primitives:
          -
            type: 1
            dimensions: [1.8, 1.5, 1.0]
        primitive_poses:
          -
            position:
              x: 0.89
              y: 0.74
              z: 0.93
            orientation:
              x: 0.0
              y: 0.0
              z: 0.0
              w: 1.0
        meshes: []
        mesh_poses: []
```

But when another component is set, this suddenly fails:
```
$ rostopic pub -1 /collision_object moveit_msgs/CollisionObject "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: 'world'
id: 'box'
primitives:
- type: 1
  dimensions: [1.8,1.5,1.0]
primitive_poses:
- position: {x: 0.0, y: 0.0, z: 0.15}
  orientation: {x: 0.0, y: 15.0, z: 0.0, w: 2.0}"

$ rosservice call /get_planning_scene "components:
  components: 16" | grep box -A20
        id: "box"
        type:
          key: ''
          db: ''
        primitives:
          -
            type: 1
            dimensions: [1.8, 1.5, 1.0]
        primitive_poses:
          -
            position:
              x: 0.89
              y: 0.74
              z: 0.93
            orientation:
              x: 0.0
              y: 15.0
              z: 0.0
              w: 2.0
        meshes: []
        mesh_poses: []
```

ggupta9777, 2018-10-25:
I'll take this up.

bdenoun, 2018-10-25:
I was  already working on it, sorry I forgot to assign it to myself :/

davetcoleman, 2019-01-14:
I'm removing the assigned tag. Did any progress ever get made on this?

j-petit, 2019-03-26:
I am taking this up. I have already found the cause in the src-files. However, I am unsure about the contribution process, as I am new to moveit!.

My environment is melodic on Ubuntu 18.04. Should I therefore make the PR to be merged into the ``melodic-devel`` branch? And should the new branch where I commit my changes be then also a copy of my forked ``melodic-devel`` branch?

rhaschke, 2019-03-26:
@j-petit, thanks for handling this issue. 
As this fix will (hopefully) not change API or ABI, it should go into melodic-devel and then be back-ported to Kinetic and forward-ported into master.
So, please branch of from the melodic-devel branch of your clone.

rhaschke, 2019-04-14:
Fixed via #1420, #1433

PR #1420 (2019-04-11): Normalize quaternions when adding new or moving collision objects (#1119)
Fixes #1119.

### Description
As proposed in #1119 I implemented quaternion normalization when adding a new collision object and also when providing a new pose (move operation). Instead of using `tf2::fromMsg` as before I wrote a new function which includes orientation normalization and replaced it in the appropriate spots.

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Created tests, which fail without this PR [reference]( [... cut]
closed 2019-04-14 by rhaschke