moveit/moveit #1120: move_group should fill in current state

v4hn, 2018-10-24:
This is a continuation of #661. Here a quick summary:

With more planners tested with MoveIt in general usage, we noticed a minor interface issue for `MotionPlanRequest`:
With the current semantics,  an empty `start_state`, as generated by the `MoveGroupInterface`, implies that the `move_group` node should assume the current state of the system as the start state for planning.
However, this is currently implemented [in the ompl interface](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L465) only and other planner plugins fail in this default use-case.
The correct start-state should thus be filled in *before* the request is passed on to the planner, e.g., by the [MoveGroup action capability](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_ros/move_group/src/default_capabilities/move_action_capability.cpp), possibly even by a new default [PlanningRequestAdapter](http://moveit.ros.org/documentation/plugins/#planningrequestadapter).

- [ ] document the current behavior in the header-comment of `move_group_interface::MoveGroupInterface::setStartStateToCurrentState()`

- [ ] provide a patch to fill in the start-state *before* the planner is called.

ipa-jfh, 2018-10-25:
I am starting to work on this as a part of WMD

ipa-jfh, 2018-10-25:
@tahsinkose you are welcome to take over this issue ;)

tahsinkose, 2018-10-25:
Yes, I'm on the exact same crossroad. An instance of a PlanningRequestAdapter is also not an automated component which is directly integrated to the planning pipeline. One who needs to use a custom plan request adapter with an external planner requires to denote additional information in launch file to as in `ompl_planning_pipeline.launch.xml`. So then, I'm continuing with non-const case.

v4hn, 2020-05-04:
The latest discussion changed the expected solution for this problem a lot.

[My comment in the attempted pull-request](https://github.com/ros-planning/moveit/pull/1165#issuecomment-442078677) summarizes my findings.

A full solution should also consider that planner pipelines can be called manually from C++ code, e.g., via `MoveItCpp`. Thus, only changing the capabilities is no full solution to the problem.