moveit/moveit #1122: Frames of collision objects

v4hn, 2018-10-24:
This relates to #1060, but the pull-request aims to do much more.

At the moment, the RViz PlanningScene Displays do not allow introspection of the coordinate systems of CollisionObjects.
It would be nice to have a way to visualize these frames in RViz.

Possible approaches include:
- conditionally publish frame information to tf and introspect with their tool
- add a bool-property tree for all objects to the PlanningScene display to enable/disable frame visualizations for them.

rhaschke, 2018-10-24:
> add a bool-property tree for all objects to the PlanningScene display to enable/disable frame visualizations for them.

+1 for this option.
Alternatively you can use the panel to select scene objects and even move them around. The orientation of the interactive mark matches the frame orientation, doesn't it?

v4hn, 2018-10-24:
Please see #1115 . We could also keep the (currently broken) axis visual there, but I guess this would be too much information in one place...
The IM should align with the frame, but judging from experience it usually does not pinpoint it very well...

rhaschke, 2018-10-26:
>  I think I would prefer publishing the frames to TF in general, so the frames are fully accessible from other nodes.

Do not agree to this. In contrast to TF, which desribes *the* current (and past) state of the (unique) world, CollisionObjects and their frames are related to a PlanningScene, which represents a possible snapshot of your world at a time. Particularly, it is possible to have multiple PlanningSceneDisplays representing different states. Publishing their transforms as TF would - in general - cause ambiguities.

felixvd, 2018-10-28:
Correct me if I'm wrong, but IIRC a move_group maintains only one PlanningScene and RobotState as its "true" current state of the world. I was picturing publishing only this "assumed true" state to TF.

That said, you are right. Planning scenes may represent possible states (although I haven't used them that way), and problems would also occur when using multiple move groups (e.g. for multiple arms). Publishing to TF would need to be optional, and should be limited to one "true" planning scene.

I do think that not being able to access the CollisionObject frames like other robot frames is inconvenient, and users will be stumped if they don't have TF-like access to them. Even simple things like checking the distance of the object tip to a target would be a pain.

What about a ROS parameter for `move_group` nodes to publish their planning scene frames? It would save writing an extra Rviz element, and this approach feels less fragmented to me.

rhaschke, 2018-10-28:
Yes, there is one "true" PlanningScene (maintained by the PlanningSceneMonitor) of move_group.
However, for planning, multiple copies of this PlanningScene will be generated.

That said, the rviz plugin doesn't have direct access to this PlanningScene C++ instance. Rather, it mirrors this PlanningScene locally within the rviz process, by subsribing to PlanningScene update msgs.

> I do think that not being able to access the CollisionObject frames like other robot frames is inconvenient.

I agree. However, it's not the task of move_group to publish them! If you want to publish your object's TFs you can simply start corresponding static_transform_publishers. IIRC, TF never forgets about transforms, even when the object is removed.

By the way, "all the other robot frames" are not published if there is no robot_state_publisher. move_group is neither publishing nor using them.

> What about a ROS parameter for move_group nodes to publish their planning scene frames? It would save writing an extra Rviz element, and this approach feels less fragmented to me.

Yes, it might be the easier approach. But it's the wrong one. Such a feature is also interesting to MTC, which uses PlanningScenes to display intermediate states. In this case it's obvious that there are multiple scenes, asking for an rviz-only extension.

felixvd, 2018-11-01:
I didn't think of MTC displaying intermediate PlanningScenes. That's a must-have feature, so I agree an Rviz-only extension will be necessary.

The argument that the robot frames are not published by move_group either is also sound. Publishing the frames should be taken care of elsewhere, but we should probably decide where. Users should not have to start their own static_transform_publishers when they add objects to the PlanningScene, in my opinion. Should PlanningSceneMonitor publish them? Or a separate node, like a scene_state_publisher?