moveit/moveit #1123: planning_scene_interface::PlanningSceneInterface needs more convenience methods

v4hn, 2018-10-24:
The `planning_scene_interface::PlanningSceneInterface` class is advertised as the standard interface to interact with the planning scene of the `move_group` node.
The purpose of the class is to simplify most changes to the planning scene to one-liners in C++.

There are a number of possible convenience methods missing though:

- [ ] `getObject` (in contrast to `getObjects`)
- [ ] `getPlanningScene` which forwards the internal `GetPlanningScene` service to the user
- [ ] `getFullPlanningScene`, which requests a complete planning scene and instantiates a `planning_scene::PlanningScene` class instance

There are many other methods that would be appreciated too, e.g., `setColor(std::string, std_msgs::ColorRGBA)`, `allowCollisions`/`forbidCollisions`, ...

Ridhwanluthra, 2018-10-25:
I am starting to work on this as a part of WMD

felixvd, 2018-10-26:
I'd be interested to know the progress on this, but I didn't see a branch in your repo. I have some changes that I did over the last month that I could add right away, including Python bindings and `allowCollisions/disallowCollisions` functions.