moveit/moveit #1124: spline-interpolation of time-parameterized Trajectories in RViz

v4hn, 2018-10-24:
The current logic to visualize trajectories in RViz only considers position information (and optionally time_from_start durations with `REALTIME`) for the displays in RViz, but the messages might contain velocity/acceleration information too.

In order to include them in the visualization, we could optionally (possibly by default) generate quintic splines from the waypoints and sample them for the visualization. This would allow a minimal level of debugging dynamics in RViz and help spot issues.

v4hn, 2018-10-24:
I agree. Explicitly displaying waypoints is more important and should remain the overall default for now.

I believe we agree that an optional spline interpolation would be nice though.
We could either change `REALTIME` mode to do this, or add another mode, e.g., `SPLINE`.
REALTIME already helps to visualize `time_from_start` problems, which *can* be
orthogonal to broken velocity/acceleration profiles.

rhaschke, 2018-10-24:
As the frame rate of rviz is not higher than 25-30 Hz, I think a simple linear interpolation should be fine as well...