moveit/moveit #1126: Create plan for improving tests

davetcoleman, 2018-10-24:
MoveIt! needs improved test coverage. The first step in tackling this is listing out what classes / packages need unit tests the most, and lack them. This is a meta issue requesting someone look into this, perhaps as a simple World MoveIt! Day task.

Related is the effort to [add a Coveralls badge](https://github.com/ros-planning/moveit/issues/1) which measures our test coverage, but it seems this is not compatible with catkin and ROS packages. I wouldn't spend time on Coveralls as several people have tried and failed with this effort already.

The outcomes of this issue:
- [ ] Create list of MoveIt! components that are good candidates for test
- [ ] Improve documentation on how to write rostests for MoveIt!
- [ ] Enlist others in tackling subcomponents

BryceStevenWilley, 2018-10-24:
Mildly relevant to this issue: it might be worth it to write test convenience utils that make it easy to make a specific robot, as a lot of tests make use of different robot models. @rhaschke has made a few effective tests that look like https://github.com/ros-planning/moveit/commit/5a1d5a8f172c43aedee9f47926068e3ac529d312, and it would be much easier to continue to write more if you didn't have to specify the entire URDF text in the test code. Some quick parser like:
```
auto robot_model_builder = ...;
// Robot looks like this:
//  a->b->c
//         \->d->e
robot_model_builder.links("a->b->c");
robot_model_builder.add_side_chain("b", "->d->e");
robot_model_builder.set_joint("a-b", "prismatic");
RobotModel robot_model = robot_model_builder.build();
```
Something like the above is much more readable and easily modifiable than URDF.

rhaschke, 2018-10-24:
@BryceStevenWilley Cool idea!
I'm afraid, that you should specify the joint type directly with the links. Something like this:
```
builder.add("a->b", Prismatic);
builder.add("b->c", Revolute);
builder.add("b->d->e", Revolute);
builder.build()
```

davetcoleman, 2018-10-24:
Yes, we def need a moveit_test_utils.cc that load a common robot model (panda?) for testing and other similar things!!

BryceStevenWilley, 2018-10-25:
I can work on this for WMD, but won't be able to get to it until 18:00 EST. If someone else wants to pick this up before then, feel free. 

Also, https://github.com/ros-planning/moveit/issues/1#issuecomment-432985241 has coverage info on which functions to test more.